area515/Photonic3D #208: Discourse forum

kloknibor, 2016-04-15:
Hi,

So I tried installing the discourse forum for some hours now... I didn't get it up the way that I wanted so I restored the backup. Installing discourse isn't that hard but discourse likes an server for its own normally... So I have to mess with proxy's to get discourse from port 80... (So I can keep hosting websites on the same server/ip)

I also dislike discourse because it's nested deep into the OS, hard to maintain, also need to install everything seperate, needs a load of resources, needs an proxy etc etc. But still I promised to let you guys test it, I'll try it again later but if someone else wants to give it a try drop me an mail and I'll send you the credentials and make an backup ;)!

Dockers runs without problems by the way :)!

Robin

jmkao, 2016-04-15:
> !DSPAM:57116e3767063044821352!

Hmm.. this shouldn't be that hard, the EXPOSE keyword in docker should
abstract the port usage inside the container from usage outside the
container. Send me the root credentials directly and I'll have a go at it
this weekend.

kloknibor, 2016-04-16:
Alright :) do you want me to install discourse and dockers already? Or do you wan to do it yourself ;)! Also give me a sign when your gonna do it so I can make an backup, I'm busy this weekend with moving to this new server ;)

Robin

jmkao, 2016-04-16:
discourse_docker (https://github.com/discourse/discourse_docker) should install as a single self contained docker container that should have no other OS dependencies. if you think you can get it done, then i can work on other stuff, but if you have trouble, let me know and i can take a look.

kloknibor, 2016-04-16:
I used this guide :

https://github.com/discourse/discourse/blob/master/docs/INSTALL-cloud.md

There is also an guide on how to add discourse first and later add al domains, but this requires more work for every domain and I don't want to get down the current domains on the server, and I don't want to much hassle to install new domains... 

So keep in mind that there are already domains online ;)

Photonic3d.com isn't moved yet to the new domain but with your hosts file you can point to the new domain to play with it ;)

Thanks and good luck! You got mail :)

Robin

Verstuurd vanaf mijn iPhone

jmkao, 2016-04-22:
We can use issue to keep track of status on this for everyone.

Discourse is installed and running with github and twitter login integration. It still needs customization work.

WesGilster, 2016-04-23:
I got a 502 bad gateway.

WesGilster, 2016-04-23:
When attempting a github login.

jmkao, 2016-04-23:
Hrmm... I see it too. Looking into it.

jmkao, 2016-04-23:
This should be fixed now, but keep an eye out for it (e.g. occasionally log out and try logging back in again with Github). I believe this is because of an flakey DNS configuration at the time I initially installed it, but I'm not certain.

jmkao, 2016-04-23:
Tasks that need to be done:

https://forum.photonic3d.com/t/read-me-first-admin-quick-start-guide/10

jmkao, 2016-04-24:
@kloknibor Can you install the following plugins for dokuwiki:

- Oauth for github: https://www.dokuwiki.org/plugin:oauth
- Image Paste: https://www.dokuwiki.org/plugin:imgpaste

I can begin populating content from there

WesGilster, 2016-04-25:
So, I got signed up and authenticated with github, but it took about 3 minutes to login. Not sure if that's normal.

kloknibor, 2016-04-25:
@WesGilster was that at the dokuwiki or at the discourse website? I'll install flarum next to this forum to compare them. I'll add the plugins today ;)!

kloknibor, 2016-04-25:
@jmkao  I installed imgpaste and Oauth, for now I only enabled facebook and github, do you guys want more options? we could add yahoo + or google+ but I don't think that's quite populair... But i can add them later today if desired.

kloknibor, 2016-04-25:
So I created an new flarum forum, all logins are fixed now ;)! And all runs smooth! forum can be found here : https://forum2.photonic3d.com/ 

So discourse : 
https://forum.photonic3d.com/

Flarum : 
https://forum2.photonic3d.com/

kloknibor, 2016-04-25:
I will try to get the menu bar of the wiki (and forum) the same as our main page so we can easily change between the pages and you won't have the feeling that you leave the website entirely ;)!

- [@kloknibor ] Changing wiki to the same looks as the main website

kloknibor, 2016-04-25:
So https://wiki.photonic3d.com is ready to go if you ask me... I made the menu just like the ones on photonic3D but I broke something... I got the good file and only made a small adjustment so I'll fix it this weekend (kingsday here so I'll be partying for some days, a good reason for a beer ;)!)

WesGilster, 2016-04-26:
That's kindof odd.  I mean, you need a reason for beer?

kloknibor, 2016-04-26:
To drink it at 8 in the morning without weird looks you need that here yes haha :p

kloknibor, 2016-04-26:
I'm working on the mobile wiki currently hope to finish it this weekend. If you guys want you can already fill up the wiki :) would be sweet if the help buttons forwarded to that, I can make that happen later ;)

Desktop wiki is fine currently, but the mobile one is a mess... It works-ish but that's about it. Will be improved soon :)

About the forum, discourse seems to be slower at first opening also most people I ask like the flarum look better (without telling my personal preference before). Also duscourse uses more resources.

Shall we make some vote thingy to choose the forum?

Robin

kloknibor, 2016-04-26:
Also I added this plugin to flarum : https://github.com/sijad/flarum-ext-github-autolink 

Think you guys would like that :)

WesGilster, 2016-04-27:
Having never used or managed forum software before, I'm finding I like flarum better due to speed and look. On the other hand, I'm not managing it though so my vote is no different than an ordinary user...

jmkao, 2016-04-27:
How would you compare the administrative capabilities, particularly around spam prevention via trust/promotion and upgrades? Spam from non-bot humans is one of the big problems in many other forums, like the buildyourownsla.com forum and the G+ 3D printing community. Upgrades will be a continuing ongoing need.

I would agree with you that in a zero content state, right now Flarum looks somewhat cleaner, but they're close enough in UI that my larger concerns are about administrative viability.

kloknibor, 2016-04-27:
Discourse runs in an docker with special port forwarding. It's nested within the server os. Flarum runs inside the domain folder and can be update with composer (update) it's pretty easy if you ask me but jmksao is the one with the more discourse experience

kloknibor, 2016-04-27:
Sorry jmkao just read your message. Flarum is easily plugina-able and currently we use akismet to prevent spam. I can set it up in such an way that every message should be improved but we could add an captcha if needed ;)

kloknibor, 2016-04-27:
Oh and updating everything including plugins is just one small command! With composer, works great if you ask me

jmkao, 2016-04-27:
How does that compare to what Discourse has? I'm really asking more for your sake since both forums now meet most of the requirements (well, I would like paste-image in Flarum since that makes posting screenshots easier, but that's a minor thing that expect to be addressed at some point and not a release requirement).

I don't actually have any discourse experience, I'm just much more familiar with direct command-line Unix system administration and Docker than I am using tools like directadmin or cPanel, so it was easier for me to configure using docker.

i'm also biased in my own way because I've run a small server farm for years that has built up 20-30 VirtualHosts and almost 50 PHP apps, many of which are custom written. I find upgrading the environment difficult because all of the different apps run in the same Apache/PHP yet have different version and library requirements. Some are continually updated to the support the newest, while others have declined in support over the years and must run with older versions of stuff. So I'm actually moving towards separating each of my apps into its own docker container and leave Apache used purely as a load balancer with proxy rewrite rules, so that each app can have its own version of its entire stack, and so that apps that are no longer supported can be left alone and be contained in their security implications to the rest of the system.

However, the functionality we are talking about here is much smaller in scope, and is not likely to increase so dramatically over time. So I'm afflicted with a certain kind of "if only had built it this way from the beginning..." kind of remorse that doesn't necessarily apply here.

Thus, putting my own personal bias aside, I mainly want to ask questions so that you can consider what you want to take on, but in the end the choice is up to you since it has been primarily your effort in starting up the sites (which we are all grateful for), and will be then primarily your effort to keep things running in the future.

kloknibor, 2016-04-27:
Sorry @jmkao I'll answer later, not in the state to give you a proper answer, I am able to make you admin and see it for yourself ;)

jmkao, 2016-04-27:
I suppose another bias of mine is that I am somewhat more afraid of projects that fail to achieve momentum than I once was. This is also not necessarily a concern for us since if we have a system that meets our requirements, we don't necessarily need an eternal stream of updates if we are wiling to put in the effort, and no project can be truly eternal, interest ebbs and flows.

What feeds my underlying bias is If you look at what seem to be the 3 most mentioned "next generation" forums, you can see their Github stats at:

https://github.com/discourse/discourse/pulse/monthly

https://github.com/NodeBB/NodeBB/pulse/monthly

https://github.com/flarum/core/pulse/monthly

Each of these projects is in a different state of their lifecycle, and you could interpret the stats differently depending on your own comfort level and perspective. For instance, discourse has no issue activity. NodeBB has a lot of issue activity, but more closed than new. Flarum has less issue activity, but more new than closed.

My bias does raise concerns for me relating to a spectrum of stability that, as I mentioned in the beginning, aren't necessarily relevant. Thus, I really want to turn this into a question as to whether you're comfortable with this, and if you are, I think we go for what you are comfortable with.

Regardless of my technical feelings around these things, my overriding project opinion is that we want to encourage people who take the initiative to make decisions and execute.

kloknibor, 2016-04-29:
So I am starting to fill up the wiki, stole some things from our github wiki and will get all to there eventually. Altough I do think some can improve but I actually quite enjoy writing the wiki... Didn't expect that haha! I'll also make some more graphical contents and some clips soon. This way it is easy for everyone to use photonic3d! I was quite new with making youtube movies and used an music track which the program I used called free, but someone claimed the copyrights so I had to change all of them, ah well just learned something new ;)!

About the forum, The community over at flarum is quite active. The stats give an bit of an wrong look at the progress from flarum. Most functionality can be added with extensions. I posted once an question about the github login and got an response within 1 minute! another pro is if something breaks that part can just get disabled ;)! And flarum is quite new. They use composer as dependencie manager and it works pretty great, it's easy to maintain and I can update everything with 1 small (composer) command! The only thing I look up against at the moment is that I need to integrate our menu bar in this forum. It is a bit harder than usual since it isn't written in php or something so I can't just add the bar and be done with it but need to write a special extension for it, but well it will be good for learning ;)!

jmkao, 2016-04-29:
Well, weigh your options,make your decision, and open the forum :)

kloknibor, 2016-04-29:
Sorry I am being hard-headed, when I'm able to make the menu-bar (Which I'm working on as we speak) we will go for flarum ;)!

kloknibor, 2016-04-29:
(It's partially personal, I love the community over there too, you can get answers on question within minutes from the helpfull people in the chat or on the forum and the Documentation is nice and fun to read ;)! So I feel more confident working on that!)

jmkao, 2016-04-29:
Don't get too caught up on customization code that isn't supported by the platform. If you like the community there, ask them for it and then move on. Writing a full custom plugin just for our header is probably not worth it to us or to them. Generalized template customization that is on par with other forum packages is likely to be something that they need to do as a generalized plugin on their roadmap. If this isn't available yet, don't get too hung up on it.

kloknibor, 2016-04-29:
The only thing I want the plugin to do is to echo the menubar code above the original header but withing the the body tags, I can echo it easily before the <HTML> but this won't be a pretty solution and will give problems. There are more people looking for the same and somebody on the chat offered to try it for me tonigh :D! So that's awesome! We will see if he succeeds, he posted another plugin he worked on and it seemed awesome too!

kloknibor, 2016-04-29:
(And I see the website as an fun challange :) I think I can move more work on the website than on the software right now but I most certainly do want to get more involved with the deeper level of the software! So I'm playing around, explore and read a lot lately ;)! Most code I write currently works but is a bit hack-ish and I want to learn program cleaner ;)!

jmkao, 2016-04-30:
@kloknibor 
Can you add https://www.dokuwiki.org/plugin:markdowku to our DokuWiki?

Also, can you investigate ways in Flarum to get RSS and image paste working? The plugin https://github.com/sthinteresting/imageattachments is supposed to allow for image paste in Chrome, but it doesn't seem to work. RSS doesn't seem to be available as a plugin, but rather some kind of manual hack to the PHP source. Might need to consult the team there what is recommended.

kloknibor, 2016-04-30:
Yes I'll add it, I tought I made your administrator to so you should be able to do it too, but I'm gonna add it tomorrow. Image pasting is added and worked fine last time I tried with flarum (I pasted the photonic3D picture after copying it from github ;)!) I'll try it again! RSS should be possible! Are you registrered? If so what is you name? I'll make your admin!

kloknibor, 2016-04-30:
For ssh I haven't made the 2 new accounts yet, I'm sorry good at pushing things forward! Jmkao you can still login with the old pssword I sended U, but I'll get everyone it's own accounts

jmkao, 2016-05-01:
Strange, image pasting is working for me with Chrome in DokuWiki, but is definitely not working for me in Flarum on either Windows or Linux.

kloknibor, 2016-05-02:
You are right Flarum isn’t sorry,  what are you looking for? Would pasting image/video’s url be sufficient? (like embed the media) or really you copied an image on an image viewer and you paste it?

From: James Kao 
Sent: Monday, May 02, 2016 1:35 AM
To: area515/Creation-Workshop-Host 
Cc: Robin ; Mention 
Subject: Re: [area515/Creation-Workshop-Host] Photonic3D Website Status (#208)

Strange, image pasting is working for me with Chrome in DokuWiki, but is definitely not working for me in Flarum on either Windows or Linux.

kloknibor, 2016-05-02:
You are right, I haven installed it properly and the extension needs an update, Looking into it!

Robin

Date: Sun, 1 May 2016 16:35:47 -0700

WesGilster, 2016-05-03:
I was fixing the Eclipse Launchers yesterday and I found that I had a pretty serious problem with my visual print material detector. Apparently it wasn't even detecting the appropriate level in my "toughSituation" test. I've fixed the test and had to dial up the tolerance to 7%. There is still tons of stuff I could do with isolating an area of interest and calibrating with circle intersections, but I felt like it's ready for use, so I placed a page out there here:
https://wiki.photonic3d.com/doku.php?id=ink_detectors

Long story short, I really like the layout of the wiki, but I think the double scrollbar is pretty 1995.

kloknibor, 2016-05-03:
Which scrollbars do you mean? Like main page and content? We could make the sidebar smaller so it only scrolls the content?

Verstuurd vanaf mijn iPhone

kloknibor, 2016-05-03:
Ah sorry, I know what you mean know, didn even notice yet ! Sorry I'll fix that!

WesGilster, 2016-05-03:
Awesome. It's starting to look good.

kloknibor, 2016-05-03:
There are different ways to handle this I initially did it this way because the menu bar would scroll away like on all other pages, but I agree double sidebars are stupid and I didn't take long posts into account.

We can handle it in the following ways : 

- add the menu bar only to the right parts of the screen and scroll away
- add the menu bar as it was but make it this way that only the content would be scrollable, the menu bar would be always visible in the wiki
- add the menu bar also just to the top and make the whole page scrollable, only keep the sidebar static
- add the menu bar also just to the top and make the whole page scrollable including the sidebar.

Currently I think I like the 3-erd option the most to fit it into the complete website but it is the most hardway too. second is quick and easy and does the job but does fit less inside the complete website and the last one has the advantage that if our sidebar gets to big it scrolls with the complete page, but to explore the wiki you always have to scroll up.

I still like the 3-erd option the most but the sidebar is concerning me on smaller displays... What is your opinion on this? It would do fine for me on a full HD screen for now... But well maybe it would be hard for some devices!

Robin

WesGilster, 2016-05-04:
My only strong opinion I have is to get rid of multiple scroll bars. At the moment there are 3 scrollbars and that's more than 1. 

Think about this from the standpoint of the user. Without drastically moving my mouse, I should be able to scroll to the proper place on the page with a single scrollbar or mouse wheel. Users aren't going to be confused when they "lose" the top menu bar, because they know they can scroll back to the top. On the other hand, they are going to be annoyed when they can't scroll their left outline menu. with the right scrollbar.

This isn't a great example of a gorgeous site, but this is what I'm talking about:
http://jbehave.org/reference/preview/

jmkao, 2016-05-04:
The behavior when there are 3 scrollbars is weird.... If you use the Window Resizer extension in chrome that will give you an idea of how the page would work on a tablet or phone. The behavior of the inner frames with the outer frame is strange and unintuitive.

One way to make the sidebar work properly on both desktop and mobile is to use a bootstrap based theme that is properly reactive (e.g. turn the sidebar into a hamburger menu when reduced), like:

https://www.dokuwiki.org/template:starterbootstrap

which would work like:

http://dokuwiki.camlittle.com/doku.php

jmkao, 2016-05-04:
Or perhaps:

https://www.dokuwiki.org/template:bootstrap3

Which seems more actively maintained and supports bootswatch themes, although I couldn't find an obvious demo site.

kloknibor, 2016-05-04:
Thanks for the suggestions! I'm gonna look into them. By the way, currently you will be switched to an special mobile template I still need to set up further when you acces the site from a none desktop device. It isn't a special mobile site or anything but it switches the sended template when an tablet or mobile device is detected :)! But well maybe that's also kind of old fashioned... I'll look into the bootstrap templates and will see What I want to do, because I do love the GUI in the wiki at the moment! Maybe I'll make it just one page so they will scroll everything down and add an " back To the Top" button, I think that will be my favourite solution, (also because it's faster to set up) but well bootstrap is cool to!

kloknibor, 2016-05-06:
I'm quite busy lately but I did a quick fix for now, maybe I have time this weekend to make the whole page scroll and keep everything where it belongs ;)! The problem was the header, it added 64px outside off the container and the container had an height of 100% so I used CSS3 Height : calc(100%-65px); to solve it for now ;)!

kloknibor, 2016-05-08:
The whole page is scrollable so that''s fine but I broke 2 things when editing the javascript :( it doesn't resize with the screen real time anymore only with a refresh and we got too much room at the bottom I can't explain yet :( Unfortunatly I don't have more time today... Will look into it later

kloknibor, 2016-05-09:
Do you think it is time to start spreading the word with the warning it's still in beta? or rather just wait?

jmkao, 2016-05-09:
We need an operational forum so that when I summon people, they have someplace to talk back to us. If that's ready to go, fix up the http/s endpoints for forum.photonic3d.com and get the categories fully configured; you can use the same ones and the same descriptive text I used in the discourse forum.

Also, I would suggest you spend less time hacking at CSS/JS directly and evaluate different template plugin to figure out the one that best meets our needs out of the box. Something that requires less line-of-code level customization will be less effort to maintain and upgrade in the future.

We're close, but we need to get one level more polished before we're ready to have people come in.

kloknibor, 2016-05-09:
Allright your right, also your right about the forums... I really do like Flarum and I think it has potential to be much better than discourse. But at the moment it's still to undeveloped, and for our use case things are going to slow... It's a waste of my time on the forum but that's my fault. 

The pasting won't happen anytime soon and I'm starting to get kind of annoyed that the updates of the forum all depend on 1 person, people can do PR's but only 1 person can approve, since he's busy this process is very slow! Also updating is better with discourse and it is just way more mature and active. Well atleast I learned something from it ;)! there aren't developed very much new extensions and only some people are active for the public domain... Also future updates may break extensions and since core functionality comes from extensions it's kind off annoying!

So for now Discourse will be a better choice... I'll play with the header soon ;)!

jmkao, 2016-05-10:
@kloknibor I see you're working on the header. One of the changes you made breaks full-screen popups, which include the upload button in a posting, and the "Show" button in http://forum.photonic3d.com/admin/logs/staff_action_logs, which makes it difficult to see what changes were made.

jmkao, 2016-05-10:
You also have a bunch of XSS blocked references from the CSS you included:

[image]

[image]

For performance and code-separation (since the Hydrogen CSS you are importing from the wiki is complex), you might try using the DOM inspector from Chrome or Firefox to look at the styling of the header and creating HTML with the same style parameters only for the header rather than bringing in the all of the stylesheets, which might bring in behavior that is harder to debug.

kloknibor, 2016-05-10:
@jmkao I was aware of these issues but didn't got more time yesterday ;)! The css files I included are way too big I can drop like 90% of the file and that's on the planning, and about the cross-origin loading I was aware too (That's also the reason the menu bar isn't working right sinds it can't find the roboto font). I was working on it yesterdat but had some issues with finding the correct symlinked folder to acces custom uploaded files to forum.photonic3d.com I found it now (in /var/discourse/share I believe at the top of my mind) but it was getting to late to fix it, sorry! The reason the fullscreen pop-ups and show button is broken is probably because of the css ;)! (The unneeded part of the CSS) so that will be easily fixable ;)!

I also tried inline css for the menu bar but that made it just worse to debug... I'll clean the CSS today and add the correct font on the subdomain, that should fix it ;)!

kloknibor, 2016-05-10:
@jmkao I broke the login button... if your still logged in can you delete the css for me? (admin -> costumize -> css/html -> header)

[19 more replies]

closed 2016-09-18 by WesGilster