area515/Photonic3D #211: [bug?] : CW code not liked anymore

kloknibor, 2016-04-16:
So I used to use this code : 

;********** Lift Sequence ********
G1{$SlideTiltVal != 0? X$SlideTiltVal:} Z($ZLiftDist * $ZDir) F{$CURSLICE < $NumFirstLayers?$ZBottomLiftRate:$ZLiftRate}
;<Delay> %d($ZLiftDist*60000/$ZLiftRate+200)
G1{$SlideTiltVal != 0? X($SlideTiltVal * -1):} Z(($LayerThickness-$ZLiftDist) * $ZDir) F$ZRetractRate
;<Delay> %d($ZLiftDist*60000/$ZRetractRate+200+$BlankTime)
;********** Lift Sequence **********

But this won't convert quite well.... : 

2016-04-16 21:33:10,677 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] Write 0: G1{$SlideTiltVal != 0? X($SlideTiltVal * -1):} Z((0.1-9) * 1) F$ZRetractRate

2016-04-16 21:33:10,678 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] lineRead: error: parse failure: can't parse Z value at line '< 4?$ZBLR:0.25'

2016-04-16 21:33:20,912 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] Write 0: G1{$SlideTiltVal != 0? X$SlideTiltVal:} Z(20.204 * 1) F{5 < 4?$ZBottomLiftRate:6.926}

2016-04-16 21:33:20,913 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] lineRead: error: line is too long

2016-04-16 21:33:20,914 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] Write 0: G1{$SlideTiltVal != 0? X($SlideTiltVal * -1):} Z((0.1-20.204) * 1) F$ZRetractRate

2016-04-16 21:33:20,915 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] lineRead: ok 1-9) * 1) F$ZRR

This used to work so what changed? I'll try to set up an custom code anyway but didn't understand why it did work and now it didn't ;)!

Robin

WesGilster, 2016-04-17:
That's never been valid FreeMarker syntax. Keep in mind, the help button will take you to the FreeMarker web site, and the "test" button usually gives really good info to help build those expressions.

kloknibor, 2016-04-17:
Alright I'll make an free marker template than :) no problem ;)! Was just wondering ;)!

closed 2016-04-17 by kloknibor

kloknibor, 2016-04-26:
What are the available variables within Photonic3D I can use to create my templates? (will forward this later to the wiki, I'll try to make 1 page a day there can't promise ;))

WesGilster, 2016-04-26:
You'll find these in TemplateEngine:
		root.put("now", new Date());
		root.put("shutterOpen", printer.isShutterOpen());
		root.put("bulbHours", printer.getCachedBulbHours());
		root.put("CURSLICE", job.getCurrentSlice());
		root.put("LayerThickness", printer.getConfiguration().getSlicingProfile().getSelectedInkConfig().getSliceHeight());
		root.put("ZDir", printer.getConfiguration().getSlicingProfile().getDirection().getVector());
		root.put("ZLiftRate", job.getZLiftSpeed());
		root.put("ZLiftDist", job.getZLiftDistance());
		Double buildArea = job.getPrintFileProcessor().getBuildAreaMM(job);
		root.put("buildAreaMM", buildArea == null || buildArea < 0?null:buildArea);
		root.put("LayerTime", printer.getConfiguration().getSlicingProfile().getSelectedInkConfig().getExposureTime());
		root.put("FirstLayerTime", printer.getConfiguration().getSlicingProfile().getSelectedInkConfig().getFirstLayerExposureTime());
		root.put("NumFirstLayers", printer.getConfiguration().getSlicingProfile().getSelectedInkConfig().getNumberOfFirstLayers());
		root.put("SlideTiltVal", printer.getConfiguration().getSlicingProfile().getSlideTiltValue());
		root.put("buildPlatformXPixels", printer.getConfiguration().getSlicingProfile().getxResolution());
		root.put("buildPlatformYPixels", printer.getConfiguration().getSlicingProfile().getyResolution());
		root.put("hostProperties", HostProperties.Instance());
		root.put("job", job);
		root.put("printer", printer);