area515/Photonic3D #212: [Feature request] : adding more profiles

kloknibor, 2016-04-16:
Hi,

I would like to be able to add more than 1 profile with the the printer import. ALso it would be nice to delete the ones present when deleting printer to update it. (just overwriting would do also).

I do love the new function by the way! I only wonder if the ink config should be in the menu underneath slicing profile... I will hide it eventually ;)!

kloknibor, 2016-04-28:
I think I will make a quick and dirty solution soon with just a clickable link to start an .sh script to empty folder and download an .zip file from the web and extract it in the folder. Will try to make it in such an way that it will only remove the files that are getting updated, this way you won't loose costum slicing profiles or different one from different printers and will try to use java.

WesGilster, 2016-04-28:
From here on out, please take a look at the API before writing something new. You are already able to upload a new version of Photonic.

kloknibor, 2016-04-28:
alright sorry! And I'm talking purely about the slicing profiles for different printers, not photonic itself. I'll take a look at the API ;)

WesGilster, 2016-04-28:
No problem at all. I thought you were talking about new versions of Photonic.

I think it would be better to add management functions(CRUD) from within the GUI.

kloknibor, 2016-04-28:
Not sure what that is but I'll find out ! (I'm an graduating mechanical engineering student, but I also like to mess around with code ;) I program PLC's currently but that's quite different than java and before I could only code some PHP,javascript,HTML,(Codesys languages ST,FL,CFC etc),python and a tiny bit of C. So it's a learning process for me! Thanks @WesGilster @ergobot and @jmkao for keeping up with me ! Your help is very appriciated!

WesGilster, 2016-04-28:
Create
Read
Update
Delete
Then add those functions into the Java API for management slice profiles and machine configs. Have you taken a look at the swagger UI? If not, you really need to. I'm pretty sure you are going to love it.

WesGilster, 2017-06-28:
This is long done.

closed 2017-06-28 by WesGilster