area515/Photonic3D #214: /printers/executeGCode only returns only "ok" w/o command response

jmkao, 2016-04-23:
The executeGCode REST API call seems to return only the last line from the printer firmware, which omits any data response for M commands.

For instance, when executing an M114 (to get the current print position), the API returns only "ok". The data part of the response is only visible in `cwh.log`

[image]

```
2016-04-23 03:10:40,360 INFO o.a.r.g.GCodeControl [qtp31345842-12] Write: M114

2016-04-23 03:10:40,663 INFO o.a.r.g.GCodeControl [qtp31345842-12] lineRead: X:0.000 Y:3.000 Z:196.000 E:0.000 Count X: 0.000 Y:2.997 Z:196.000

2016-04-23 03:10:40,666 INFO o.a.r.g.GCodeControl [qtp31345842-12] lineRead: ok
```

WesGilster, 2016-04-26:
If you want to test this, and see if it works to your liking, I'll promote our build to production.

WesGilster, 2016-04-26:
Wait first, we need to delete start.sh and newstart.sh to start.sh correct?

jmkao, 2016-04-26:
You can still check in the cwh-0.XXX.zip as before, but I think that newstart.sh seems ready to go.

The only difference for prod is do not use the `fullRelease` gradle target. Just push a tag and CI will take care of the release.

jmkao, 2016-04-27:
Cool, it works well. Looks GTG.

[image]

closed 2016-04-27 by jmkao