area515/Photonic3D #216: Backwards compatibility

WesGilster, 2016-04-29:
I thought we had backwards compatibility with the old start.sh. As I attempted an install. It wiped out my CWH directory and left me with nothing again. Is everyone going to have to install this on their own?

WesGilster, 2016-04-29:
Here are the logs from my start

pi@raspberrypi /opt/cwh $ sudo ./start.sh
mv: cannot stat `cwh-0.*.zip': No such file or directory
--2016-04-29 01:49:17--  https://github.com/area515/Creation-Workshop-Host/raw/master/host/build.number
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/build.number [following]
--2016-04-29 01:49:20--  https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/build.number
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 23.235.44.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|23.235.44.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 80 [text/plain]
Saving to: `build.number'

100%[======================================>] 80          --.-K/s   in 0s

2016-04-29 01:49:24 (9.70 MB/s) - `build.number' saved [80/80]

Installing latest version of cwh-0: 298
--2016-04-29 01:49:24--  https://github.com/area515/Creation-Workshop-Host/raw/master/host/cwh-0.298.zip
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-04-29 01:49:27 ERROR 404: Not Found.

unzip:  cannot find or open cwh-0.298.zip, cwh-0.298.zip.zip or cwh-0.298.zip.ZIP.
chmod: cannot access `*.sh': No such file or directory
rm: cannot remove `cwh-0.298.zip': No such file or directory
Turning off screen saver and power saving
Determinging if one time install has occurred
Starting printer host server

WesGilster, 2016-04-29:
I don't really have any way to test this since I "used up" my last install of the old version of CWH.

jmkao, 2016-04-29:
We need to keep 1 copy of the zipfile that matches checked-in `build.number` for the old version of `start.sh` to work.

jmkao, 2016-04-29:
I'll fix and test this along with some binary cleanup. I think I have the old startup.sh on the Raspberry Pi 1 I was using to test the image.

WesGilster, 2016-04-29:
I see. Sorry about that, I should have left you create the tag since I wasn't quite sure how this fits together.

kloknibor, 2016-04-29:
For testing purposes we could use an old image right? That one got CWH preinstalled, maybe we would need need the start script in the image to not update the software ;) This can be easily done with OSF mount and an linux partition viewer from windows.

jmkao, 2016-04-29:
For testing the old script, we would not be able to make any changes to it, since migrations from old versions is the point. I'll test it after my 10 hour flight home.

kloknibor, 2016-04-29:
hmm alright, I do have 2 old installs at home, will test it later. And have a good flight @jmkao !

jmkao, 2016-04-29:
Actually, stupid brain freeze. I can easily get the old start.sh from the last tag in git.

jmkao, 2016-05-01:
This works now.

There is a `cwh-0.298.zip` checked in (via LFS so it doesn't actually take up any space in our repo) that will bridge existing users from the old `start.sh` to the new `start.sh`.

The presence of this one-time file in the repo means that people have to be careful not to erase this file when checking stuff in. It also means that the build.number field in `build.number` must not increment, otherwise the logic in the old `start.sh will not match 298.

I have updated BuildCWH.xml to generate zip files always by tag name or check-in hash, which was previously only used when CI does the build. That means that when you do a `./gradlew fullRelease`, the name of the zip created will include the tag you enter at the prompt rather than the incremented number. If you manually create a dist zip via `./gradlew dist`, then the suffix will be the hash of the current commit.

In the long term, this should make things easier since tags, being strings, can be used to identify whose repo a build was generated from, if we use non-overlapping tags for our own repo releases. Compared to build.number, which might overlap. This should be a net benefit now that I have altered the way that Travis CI works so that anyone can do CI driven automatic releases, not just area515.

In the short term, this means getting used to a more identifier that is more tied to git than time. It also means that we need to update the UI to use the value of `repo.version` instead of `build.number` for the display in the UI. I will work on this tomorrow.

Also, the easiest way to prevent from accidentally disrupting the cwh-0.298.zip left for backwards compatibility is to run `git update-index --assume-unchanged cwh-0.298.zip` when in the `host` directory of your local repo. Then, your local repo will ignore whatever happens to this file, so if you delete it, your deletion won't get checked in. Unfortunately, this can only be done on the client side.

I'll also make a change to `build.gradle` that will special case exclude this file from `./gradlew clean`.

closed 2016-05-02 by jmkao