area515/Photonic3D #218: Rename or delete and recreate

WesGilster, 2016-05-01:
So I'm trying to do some compares from older versions of my algorithm and everything is telling me that my code doesn't exist in older versions. When you performed the code move from src to src/main/java, did you do a git rename on the source directories or did you create a new directory and create "new" files from the source and check those in. All history seems lost for all source files.

jmkao, 2016-05-01:
I definitely renamed them with `git mv`. Let me look into it

jmkao, 2016-05-01:
Hmm that is weird...

Commit 7fe642d1 show that the files were "renamed without changes", but their history seems to have gotten detached from the files.

If you browse the tree at the commit directly preceding that one (https://github.com/area515/Creation-Workshop-Host/tree/3558a18c1cc23761d64fac943bfc5aa52ae7f3e7/host/src/org/area515/resinprinter/server), the files have the correct history.

But the files after that point do not.

jmkao, 2016-05-01:
Ahh, I see. File renames in git are not as powerful as I thought they were:

http://stackoverflow.com/questions/2314652/is-it-possible-to-move-rename-files-in-git-and-maintain-their-history

In particular, history is not continuous unless you do the hack of rewriting prior commits using `git filter-branch`, which after the repo cleanup, we're now well acquainted with. Instead, a pointer is added that git tooling can follow.

For instance if I do `git log ZSclier.java`, I get:

```
$ git log ZSlicer.java
commit 97ee22ea60bb7468c5eac950c64d4f5a17a15f8a
Merge: 7149fa5 7523446
Author: James Kao <jmk17@cornell.edu>
Date:   Mon Apr 25 09:43:45 2016 +0200

    Merge latest WesGilster branch changes with gradle reorg
    
    
    Former-commit-id: c4045d43ec64bdb9f251295ae5bf979c1b73c47d

commit 33667cf8306cedb5f1bb607893999a605a97d118
Author: James Kao <jmk17@cornell.edu>
Date:   Sun Apr 24 15:39:45 2016 +0200

    Moved src and test files, separating out resources, per gradle standards, reducing the custom build configuration down to just the conf files
    
    
    Former-commit-id: 7fe642d1e45bded0e75b1732b73e62547fb1d8ce
```

But if I use --follow, then `git log --follow ZSclier.java` can get the history of the file across renames:

```
commit 33667cf8306cedb5f1bb607893999a605a97d118
Author: James Kao <jmk17@cornell.edu>
Date:   Sun Apr 24 15:39:45 2016 +0200

    Moved src and test files, separating out resources, per gradle standards, reducing the custom build configuration down to just the conf files
    
    
    Former-commit-id: 7fe642d1e45bded0e75b1732b73e62547fb1d8ce

commit f5fa109f430d75b2efecb7dc924ead8692423f20
Author: wgilster <wesgilster@gmail.com>
Date:   Sun Apr 17 01:43:49 2016 -0500

    Added a slice offset
    Added ability to watch y scan lines while rendering
    Added ability to record pixel tests to a json file for later execution
    Improperly computing normals when mending broken loops
    Stop rounding, triangle zintersection code
    
    
    Former-commit-id: dd8e5627cdd51b3d694ff0c9b133e6eb436777c4

commit 34903dd3e19d20d66677461030edb3d386e9710a
Author: wgilster <wesgilster@gmail.com>
Date:   Sat Apr 9 08:27:08 2016 -0500

    Fixed huge bug in geometry cutting
    Added debugging to the slicer when in eclipse
    The slicing algorithm will now allow you to trace Triangles that you
    have selected in the SliceBrowser through the next slice.
    Refactored the SliceBrowser a bit.
    Refactored the SliceBrowser a bit.
    STLErrors have a nice toString() now.
    Silenced a todo log statement
    
    
    Former-commit-id: 5155f49060114da8585b3673c6e0e9249d03fe2b

commit 338d5e0131cfef5c2ddc6c101d05324aa5ba9f2a
Author: wgilster <wesgilster@gmail.com>
Date:   Sun Jan 24 11:55:20 2016 -0600

    cwh-0.207
    Logging rewrite to high performance log4j2
    Utilized message format printing and occasional closures to gain higher
    performance than what sysout can do alone

...
```

But the behavior means that the nice Github UI history browser does not work based the rename point, since it doesn't seem to have the capability to traverse a rename like `--follow` does. The closest workaround I can think of for now is to browse the tree at the prior commit:

https://github.com/area515/Creation-Workshop-Host/tree/3558a18c1cc23761d64fac943bfc5aa52ae7f3e7

This will give you history visibility at that point and prior.

Sorry about that, the semantics of rename surprise me as well and I didn't realize that we'd be creating a discontinuity by doing that. I guess the last week of April in 2016 will be an extra special point for our repo. I'll add a section to the April 2016 wiki page to note this for history comparison.

jmkao, 2016-05-01:
It seems that the Eclipse git history browser can traverse renames, although with some limitations:

[image]

Eclipse will show a history of commits that affected the file that traverses the rename, but its comparison tool can only do "Compare with Workspace" after the rename. For versions of the file before the rename, you can only do "Compare with Previous" (which interestingly enough, can compare across the rename, but is incapable of going more than 1 version at a time).

[image]

WesGilster, 2016-05-01:
I see. It's not your fault James, the best we can do is a git rename. The one thing I can say is that you are always willing to dig in deeper, I really appreciate that. I'm certainly not a git expert and apparently this is the first time I've had to view history through a rename. I thought the tools would allow this to be more transparent. Thanks for the help!

closed 2016-05-14 by jmkao