area515/Photonic3D #220: fullRelease uploads the backwards compatibility zip to releases

jmkao, 2016-05-09:
The gradle `fullRelease` target uploads all of the zip files in the cwd to the release, which picks up the LFS stub cwh-0.298.zip in addition to the cwh-TAG.zip.

This is confusing for users who are downloading the zip manually, and also confuses start.sh.

Need to fix start.sh to be more specific about downloading the release matching the tag (which is the "new world" behavior) and to fix `fullRelease` to omit the backwards compatibility zip.

WesGilster, 2016-05-21:
Quick question on this. I tried to release a dev version, and it didn't promote my release to the "latest". Any idea why that happened?

jmkao, 2016-05-23:
I believe github determines the latest release based on the date of the commit that is tagged for the release. It looks like DEV002, photonic-dev003, and photonic-dev004 are all on commit ffda795, which makes Github think that all of those releases are the same. What seems to happen is that when a release is on a newer commit, that release will become the latest. Then if you release again on the same commit, the release will be added, but will not become the latest. I guess this is some kind of comparison rather than sort based behavior.

WesGilster, 2016-05-23:
I see that makes sense. Thanks for the help. I was just trying to get rid of a build with a misspelled tag name photomic... Next time I'll check something in before a build to make sure the "latest" is taken.

closed 2016-05-23 by WesGilster