area515/Photonic3D #222: Can't set name for slicing profile supplied with JSON config to /printers/save

jmkao, 2016-05-31:
I'm in the process of creating a JSON printer template for the Kudo3D Titan 1, and am trying to set the slicing profile's name to something descriptive about the coefficients that I set (e.g. vat type and resins in perhaps an easy/medium/hard categorization).

It appears that no matter what you set for `slicingProfile`->`name` or the `SlicingProfileName` fields, the actual slicing profile created will always be the same name as the printer.

I traced this in the browser and verified that it was POSTing the payload retrieved from https://raw.githubusercontent.com/jmkao/Photonic3D/master/host/printers/Titan1.json to /printers/save correctly. I also noticed that in the `mUVe 1.json` file, there is a `SlicingProfileName` field set to "Make Juice SF" that also seems to not manifest in any of the resulting created files.

I can see the matching fields in the `SlicingProfile` and `PrinterConfiguration` classes and followed the logic to `HostProperties.saveConfigurations()`, but that method seems to name the file via:

```
File slicingFile = new File(PROFILES_DIR, currentConfiguration.getSlicingProfileName() + PROFILES_EXTENSION);
```

Which would appear to use the value set in SlicingProfileName, but somehow the actual files created are named with the machine config name instead.

WesGilster, 2016-06-03:
Sorry didn't see this bug. Fixed now. This was a throwback to when it wasn't easy to control how a printer was saved through different client situations. Eventually we need to bring in protractor to start getting tests into the front end...

jmkao, 2016-06-10:
Whoops, forgot to update the issue. This works now. Thanks!

closed 2016-06-10 by jmkao