area515/Photonic3D #226: Solidworks STL files return an error

kloknibor, 2016-06-13:
Hi,

I've got a small bug currently because Solidwork saves the files different than most stl files. 

I get the following error : 

Failed Format Error:expecting 'facet' on line 1

So I checked the stl file and found out SolidWorks adds "Solid $nameofthepart" in the first line of the stl file and Photonic3D can't handle this.

log bundle :
[LogBundle (3).zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/311395/LogBundle.3.zip.pdf)

Stl file : 
https://www.dropbox.com/s/grt4wjb03ppav72/lager%20houder.STL?dl=0

I think this can be easily solved with if no facet found on first line try second line kind of way. But maybe there are better suggestions :)

WesGilster, 2016-06-16:
This is fixed and checked into my branch with a test for this situation as well.

kloknibor, 2016-06-16:
This seems fixed :)! works well once again!

closed 2016-06-16 by kloknibor