area515/Photonic3D #227: ZSlicing Geometry Tests Run Out of Memory When Heap is Limited to 1GB

jmkao, 2016-06-15:
The gradle test task limits heap to 1GB. On Windows I'm seeing that two of the tests run out of memory currently:

```
org.area515.resinprinter.test.FullTestSuite > org.area515.resinprinter.stl.ZSlicingGeometry.testSpecialPoints[
0] FAILED
    java.lang.OutOfMemoryError: Java heap space
        at org.area515.resinprinter.slice.StlFile.readBinaryFile(StlFile.java:318)
        at org.area515.resinprinter.slice.StlFile.load(StlFile.java:408)
        at org.area515.resinprinter.slice.ZSlicer.loadFile(ZSlicer.java:937)
        at org.area515.resinprinter.stl.ZSlicingGeometry.<init>(ZSlicingGeometry.java:39)

org.area515.resinprinter.test.FullTestSuite > org.area515.resinprinter.stl.ZSlicingGeometry.testPointsOnAllSli
ces[0] FAILED
    java.lang.OutOfMemoryError: Java heap space
        at org.area515.resinprinter.slice.StlFile.readBinaryFile(StlFile.java:318)
        at org.area515.resinprinter.slice.StlFile.load(StlFile.java:408)
        at org.area515.resinprinter.slice.ZSlicer.loadFile(ZSlicer.java:937)
        at org.area515.resinprinter.stl.ZSlicingGeometry.<init>(ZSlicingGeometry.java:39)
```

WesGilster, 2016-06-15:
This is a problem, it's fixed and I've checked it in. These heap issues happen when ASCII files are interpreted as binary files because the face count is interpreted as a binary int from what was designed to be 4 printable ASCII bytes. The smallest binary int you can possibly build from 4 printable ASCII chars is four spaces which ends up being 2.5 gig triangles. Well above the established 1 gig limit.

I alluded to this before in some other bugs, but didn't really describe this in any detail. I describe it in detail because it's understandable that memory issues can be concerning on embedded devices.

The solution I provided isn't permanent, and will be fixed when #226 is fixed.

WesGilster, 2016-06-17:
More permanent fix applied and released into production

closed 2016-06-17 by WesGilster