area515/Photonic3D #228: Bug : 3d viewer not slicing well

kloknibor, 2016-06-16:
I just found out that the test file from #226 accidentally has screwed sizes. I should have drawn in mm but it appears it became metres. 

The result is that the 3D viewer doesn't scale properly or does this depend on the printer area?

Stl file :
https://www.dropbox.com/s/grt4wjb03ppav72/lager%20houder.STL?dl=0

3D viewer shows black because it's inside the part and when moved I only can see 2 traingles stretching out of the screen.

WesGilster, 2016-06-17:
The real problem is that your model is too big. Unless otherwise specified by an externally defined option, the scale of an STL is always 1.

Two things can happen, both are one line of code. I can change the camera position to the topmost point of the printer area. That would work for most models, however given the size of your STL, that might still put camera inside your model. The other option is that I can put the camera a certain distance (percentage of the models size) away from the model. I don't like this option as much because it causes all models to "seem" to have the same scale no matter how large they are in comparison to the build area.

jmkao, 2016-06-17:
I think that make the camera position some point relative to the build area makes more sense than making it try to always fit the model. A model that's too big to fit in the build area can't be printed anyway, and I think between previewing a print and previewing the model, the print is the better alternative.

I think about this from the perspective of an inattentive user. If someone loads an STL where the unit scale is off, if the preview looks like just a few triangles, they'll think that something is wrong and not try to print it. We could incrementally improve things with warnings that detect conditions like "warning: you object is 5000mm wide and doesn't fit in the build area" in the future.

On the other hand, if the camera is too smart and focuses on previewing the model, then the user might accidentally try to print this and then there would be all kinds of downstream ramifications, like someone thinking "I opened an STL that looked okay but then Photonic3D made my printer try to print a solid cube which damaged my vat and consumed $50 of resin!" when in reality it's user error.

kloknibor, 2016-06-17:
I think the first option, put the camera in the corner of the printer area, would work best and eventually notify the user that the parts is too big for the printer area.

kloknibor, 2016-06-27:
Hi Wes!

I finished up my courses and it helps understanding the code quite a bit better :)! 

Yes I see how `loadModel(newValue);` is being called and that it will use this : `camera.lookAt(currentMesh.geometry.boundingBox.min)`; for the camera.

I think I should change the scene declaration to : 

```
//Scene and camera setup
						scene = new THREE.Scene();
						camera = new THREE.PerspectiveCamera(45, width / height, 1, 1000 );
						camera.position.x = printJob.printer.configuration.machineConfig.PlatformXSize;
						camera.position.y = printJob.printer.configuration.machineConfig.PlatformYSize;
						camera.position.z = printJob.printer.configuration.machineConfig.PlatformZSize;
						camera.up = new THREE.Vector3(0, 0, 1);
```

and add the following to load model : 
```
var width = getWidth();
						var height = getHeight();
camera = new THREE.PerspectiveCamera(45, width / height, 1, 1000 );
						camera.position.x = 80;
						camera.position.y = -80;
						camera.position.z = 300;
```

Am I on the right path?

WesGilster, 2016-06-27:
Looking great! Now test your solution out with a few models and you'll start to notice a problem that starts to merge into James's AffineTransform fix. Model designers rarely take in consideration where in 3d space they are building their model. That means they might be building their model with the center point of their model base at 5000000x400x3000. For our 3d viewer, that's not really a problem since we "lookAt" boundingBox.min for the model. However, what happens when you try to draw a simple cube around the model representing the printers build area? In fact, it looks like I already have a method that does that. Why did I comment out the method call?

kloknibor, 2016-06-28:
Thanks for your answer! And yes I found your method for drawing an bounding box and axis! Was already wondering why it was commented out ;)! I will test the code change when I got time and will see what happens :)!

Thanks for walking me through this! It's appriciated :)!

WesGilster, 2016-09-18:
Closed in cwh-1.0.0-beta3.

closed 2016-09-18 by WesGilster