area515/Photonic3D #229: Feature request : Add more than 1 slicing profile when adding printer

kloknibor, 2016-06-16:
Hi,

I think it would be great if we could download more slicing profiles depening on the printer chosen. So if we choose photocentric LC 10" that all files will get downloaded to the slicing profile folder.

I think easiest is to make a folder just like printers (or inside printers) with sub folders of with names of the supported printers and pull the profiles from there.

Let me know what you guys think.

WesGilster, 2016-06-17:
Yeah, that sounds great, but I'm looking at doing something a little bit more cohesive than throwing everyone's data into a global location like github.

kloknibor, 2016-06-17:
Ahh okay, I wouldn't think that slicing profiles would matter. Feel free to use the server if needed!

WesGilster, 2016-06-17:
Thanks, it's not really a question of storage. Every host already holds it's own data, it's more a question of computing and IO right now.

WesGilster, 2017-06-28:
I believe you have already solved this bug Robin. We are now able to create multiple slicing profiles and ink configs. Will close if you have no objections.

closed 2017-07-11 by kloknibor