area515/Photonic3D #230: Piwik

kloknibor, 2016-06-16:
I inserted Piwik inside our website, if somebody else is intresting in keeping track of the website stats let me know and I will create a login for you.

I did spread the word about photonic3D a tiny bit, most people report that it's still a bit complicated but well it's still beta and I warned them ;)!

we got 32 views in the last 24 hours and got 5 downloads in the last 40 hours.

closed 2016-06-16 by kloknibor

WesGilster, 2016-06-16:
Interesting. I think we need to work on that complicated part... More community managed printers would be better. Kloknibor, as an related side note, I noticed that your Photocentric printer configuration still doesn't have working templates. Have you noticed that?

kloknibor, 2016-06-16:
hi Wes,

Feel free to call me Robin :) And yes I noticed, but since no printer templates work currently because of #231 I haven't updated it... 

And yes we should work on making this more plug and play. Maybe release a GUI focussed on a 1 printer environment since most people will use this for an pi anyway. This way you don't have to make complicated steps to reach printer controls etc.

also I think we should more work in the flow users will make. 

Most of the time the pages you are most intrested in when you open the software is printables or current printjob. If nothing is started you are most likely intrested in the current printables and their transforms. If you are printing you will be intrested in the current printjob.

I think we should be good with 2 tabs and a setting pop-up for an 1 user environment. 

Integrate the printer controls inside the current printjob and make it 1 page max which is aligned well.

printables just be a list with printables and next to it an 3D viewer with options to do the wanted transforms and object placement. When done you can send it to the printer, get a pop up to ask to slice with what machine settings (if there are more than 1) and what slicer settings and voila. 

Configuring the profiles and printers can be hidden in an settings pop-up

Let me know your thoughts.

jmkao, 2016-06-16:
If you are hearing feedback from people, encourage them to post requests in our forum. Getting forum participation up will give us more routes to solicit feedback.

In particular, for feedback about the UI, it would be good to have a concrete list of complaints to fill a pipeline of improvements and start a discussion. This is something I want the interns to work on, but I need concrete project ideas.

kloknibor, 2016-06-17:
I already am trying to get the developers between them to use github and all normal users with feedback and questions onto the photonic3d forum. Despite some people start to PM me instead of posting here or there. I will try to get them to post on one of the forums!

WesGilster, 2016-06-17:
I wouldn't worry about creating a brand new GUI to focus on these enhancements, it's really not worth the fork. Ultimately forks of the GUI are when you are looking to have deeply integrated features,  crosscutting features, or a drastic change in technologies. I don't really see anything here that hints towards those kinds of changes.