area515/Photonic3D #231: Addint printer from github broken

kloknibor, 2016-06-16:
Hi,

When you try to add an printer from github to photonic3d it will return an server error 400. (400: Invalid request )

was using it from windows and is reproduceable, logs can be found here :  

[LogBundle(1).zip](https://github.com/area515/Creation-Workshop-Host/files/317971/LogBundle.1.zip)

jmkao, 2016-06-16:
Hmm, strange, in the Chrome inspector, it seems like there is a 400 error coming from POSTing to the URL that contains the printer JSON.  Trying this out in Postman directly, GETing the URL works fine as expected, but POSTing to it will result in a 400 error. However, I'm not sure what the logical flow of this section of code is since the JS stack trace only goes up to a JQuery call.

[image]

WesGilster, 2016-06-16:
Right off hand, this looks like a CORS issue. I saw a commit a while back for a fix to a CORS issue. Did that get fully resolved?

jmkao, 2016-06-16:
If you're referring to da2ce0f53a7bd3c107748f15e236d8ada56e077f, that specifically had to do with CORS restrictions on following 302 redirects for repos renamed to Photonic3D (specifically, mine).

But CORS issues show up pretty explicitly in the browser console. This seems to be something different related to the HTTP method used.

WesGilster, 2016-06-17:
I quick checked this out by replacing the POST request with a GET and the preflight CORS options request fails. However the POSTman request and a basic browser request work fine.

WesGilster, 2016-06-17:
It's possible github tightened their CORS restrictions. I check into this tomorrow. 

I'm really second guessing if Github is the best place to be keeping these resources anyway. I've got bigger plans for how all of these resources should be shared with one another. I just don't have all of the design work done to replace it with that yet.

WesGilster, 2016-06-17:
It looks like they are now using CORS to ensure people use download urls only from their site:
default-src 'none'; style-src 'unsafe-inline'

However for small enough files, apparently we can still snag them through their restful API. Checked in a fix.

jmkao, 2016-06-17:
I tested this and it's working now.

Between this and the start.sh fix that I just sent to you in a PR, how close are we to generating a new stable release? Stable's printer settings download is probably also not working.

I tried to cherry-pick just the relevant commits, but it's tricky because for changes to printers.js which has other changes in it that are tied to server changes as well.

kloknibor, 2016-06-17:
Feel free to use the photonic3d server to actually pull the data from!

WesGilster, 2016-06-17:
Merged and pushed a new production release.

kloknibor, 2016-07-03:
This bug is back! I can't seem to download any printers from github and I can't reopen this issue.. Sorry

kloknibor, 2016-07-12:
https://www.youtube.com/watch?v=X41m_nDrS-g&feature=youtu.be Here you can see it, for some reason the area515 one closes right away. This wasn't the case before... This was actually the first time I encountered this! So windows version is broken I suppose for now... But when I run from the gradlew run command it does run, but you will see the error 400 and not downloading templates

WesGilster, 2016-07-12:
In order to determine why it closes immediately you'll need to run start.bat from a command prompt instead of just double clicking on the icon. As you can see, when you just double click, it doesn't tell you anything.

As to the other issue, it seems there is either a bug in the repo that you are using, or you have ill formatted json printers. My bet is bad json, but I can't really tell without logs.

WesGilster, 2016-07-12:
Downloading and running Release 1.0.0-beta2 works great for me and adding printers also work fine.

kloknibor, 2016-07-13:
[image]

Not much more info when running from the command prompt

but judging from : 
[log.err.txt](https://github.com/area515/Creation-Workshop-Host/files/362444/log.err.txt)

My java set-up is broken. (which is funny because compiling works still fine + I didn't temper with the java settings)

WesGilster, 2016-07-13:
Unfortunately, this isn't a problem on the Java end. It looks like a browser issue. Since I can't reproduce this and the browser does have logs, we are in quite a pickle. You'll need to open the chrome integrated debugger(ctrl+shift+i) and look at the network traffic that looks like this:
[image]

There should be a highlighted red resource that is giving you the 400 message. When you find that, you need to take a look at what the response says.

kloknibor, 2016-07-14:
I understood the printer not downloading from github wasn't a java bug, but the CMD not starting was :)

But after reading your comment and taken an shower, I was thinking about your comment and thought, Shit! Not again... It might be the stupid Ad Blocker I use! It won't be the first time I wasted time by finding out where elements went only to find out it was the fault of the adblocker....

So I just tried it in incognito mode with chrome and it works like a charm! Sorry for wasting time! At least now we now the adblocker can act up! I will close this now.

Thanks for your help

closed 2016-07-14 by kloknibor

WesGilster, 2016-07-14:
It sounds like the CMD wasn't a problem with Java either, it was WinAVR manipulating the state of your system without your knowledge. The box that you are testing with sounds like it's in a constant state of flux. Until you stabilize the software on that box, you are really going go through some pain.

Eventually you may run into a legitimate issue that can only be reproduced periodically. If you don't have a stable system and the software on your box is constantly changing, you'll never track down what is going on. As James mentioned on that imaging thread, it's important to track down specifically the one factor of that changed on your image. Thanks again for the testing, I'll keep the adblocker in mind.