area515/Photonic3D #232: Cannot download software from github using wget

camhguh, 2016-06-16:
connected

Error msgs 
gnutls: A TLS fatal alert has been received
gnutls: Received alert [0]: close notify

Unable to establish SSL connection

jmkao, 2016-06-16:
Can you try it again? Occasionally I have found there can be an intermittent connection failure that goes away if you retry.

If that doesn't work, can you paste the full output of the `wget`, including the full command that you are entering.

camhguh, 2016-06-16:
Tried several times but will try again :)

camhguh, 2016-06-16:
sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh 

was the wget command that I have been using.

camhguh, 2016-06-16:
Still the same error which Google suggests is an SSL certificate problem?

jmkao, 2016-06-16:
That would be strange because I can hit the URL fine a a web browser like google chrome. Can you paste the full output of the wget command, especially any redirections that occur so I can see if the path you're getting is any different than what I get?

camhguh, 2016-06-16:
This is the output from PuTTY

pi@raspberrypi:~ $ sudo wget https://github.com/area515/Creation-Workshop-Host/r      aw/master/host/bin/start.sh
--2016-06-16 15:44:41--  https://github.com/area515/Creation-Workshop-Host/raw/m      aster/host/bin/start.sh
Resolving github.com (github.com)... 192.30.252.121
Connecting to github.com (github.com)|192.30.252.121|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/area515/Creation-Workshop-Host/maste      r/host/bin/start.sh [following]
--2016-06-16 15:44:42--  https://raw.githubusercontent.com/area515/Creation-Work      shop-Host/master/host/bin/start.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 62.252.172.24      1
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|62.252.172.2      41|:443... connected.
GnuTLS: A TLS fatal alert has been received.
GnuTLS: received alert [0]: Close notify
Unable to establish SSL connection.
pi@raspberrypi:~ $

kloknibor, 2016-06-16:
Hmmm I got this before and found somewhere that this is more common with github, you could add a line to ignore the ssl error and just download.

In my case I eventually found out I was doing something stupid like having no internet connection to the pi or something or something with the firewall, not sure anymore. Well try to remember what it was, in the mean time I'll find the ignore ssl error command ;)!

camhguh, 2016-06-16:
Thank you! 
I am not familiar with Linux at all so I would need some help in constructing the command.
I'm assuming (dangerous I know) that the pi internet connection is ok because it registers as connected.

WesGilster, 2016-06-16:
You could try the solution provided here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=131570

kloknibor, 2016-06-17:
After further searching I found out my error was an other one, it was missing an CA certificate back than and isn't the same as you encounter. The first thing i found on google was also what was mentioned, could you try that and let us know?

PS what the solution there is, is changing the DNS server. To do this from command line try the following steps : 

Connect with ssl to your pi.

enter the command : nano /etc/resolvconf.conf

(you might neet root for this file not sure, if so use sudo before the command like : sudo nano /etc/resolvconf.conf)

find the line : "name_servers" just use your arrows. And before that line there will be # remove the "#" and enter after the name servers : 8.8.8.8 

I hope that this simplifies what you need to do! Please let us know how it works out for you.

kloknibor, 2016-06-17:
@camhguh also if this doesn't work you can try our special image for the pi from here : https://photonic3d.com/downloads/

But unfortunately the pi 3 has a problem, see #175 I opened a while ago. You should enable device tree and your wifi will get stable! Before that you better use a keyboard to enable device tree.

camhguh, 2016-06-17:
Same error sadly,
 i'm using ethernet cable connected from laptop to the pi, not wi-fi.

will try the special image :)

# Configuration for resolvconf(8)
# See resolvconf.conf(5) for details

resolv_conf=/etc/resolv.conf
# If you run a local name server, you should uncomment the below line and
# configure your subscribers configuration files below.
name_servers=8.8.8.8

# Mirror the Debian package defaults for the below resolvers
# so that resolvconf integrates seemlessly.
dnsmasq_resolv=/var/run/dnsmasq/resolv.conf
pdnsd_conf=/etc/pdnsd.conf
unbound_conf=/var/cache/unbound/resolvconf_resolvers.conf

$ sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
--2016-06-17 09:12:03--  https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
Resolving github.com (github.com)... 192.30.252.128
Connecting to github.com (github.com)|192.30.252.128|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/bin/start.sh [following]
--2016-06-17 09:12:04--  https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/bin/start.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 62.252.172.241
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|62.252.172.241|:443... connected.
GnuTLS: A TLS fatal alert has been received.
GnuTLS: received alert [0]: Close notify
Unable to establish SSL connection.
pi@raspberrypi:~ $

kloknibor, 2016-06-17:
That's a pity!

I also own a raspberri pi 3. (like all other raspberri pi B generations :P)  I will try to get it working on this device too, I hope/think to have some time this weekend, and ethernet should work fine!

kloknibor, 2016-06-17:
I just tried the wget from a pi 3. No problem at all but we (well @WesGilster ) have been playing around to fix download issues from github, you can try it one last time. For some reason the image didn't work properly but it was one build be me so I will investigate that further, might be I inserted ethernet too late :P

WesGilster, 2016-06-17:
This is some sort of networking config with his Pi. Unfortunately, my fix for the printer config install won't affect this. I'd like to see if wget will work on several urls, or if he's just having trouble with wget and github.

camhguh, 2016-06-17:
Will give it another try later, I do battle with a chronic illness so I have to wait for a period where my energy levels are good and I can concentrate :)

kloknibor, 2016-06-17:
Wes we could do an teamviewer session so you can play with the pi3. Just tell me what you need prepared ;)! Also I got an pi touchscreen today. I haven't succeeded to get life in it but working on it, seems I'm not the only one!

camhguh, 2016-06-17:
yes....looks like I got a download :)

pi@raspberrypi:~ $ sudo  wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
--2016-06-17 20:38:21--  https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
Resolving github.com (github.com)... 192.30.252.122
Connecting to github.com (github.com)|192.30.252.122|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/bin/start.sh [following]
--2016-06-17 20:38:22--  https://raw.githubusercontent.com/area515/Creation-Workshop-Host/master/host/bin/start.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.31.17.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.31.17.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5946 (5.8K) [text/plain]
Saving to: ‘start.sh’

start.sh            100%[=====================>]   5.81K  --.-KB/s   in 0.001s

2016-06-17 20:38:22 (10.2 MB/s) - ‘start.sh’ saved [5946/5946]

kloknibor, 2016-06-17:
Great! And did you got it to install?

camhguh, 2016-06-17:
Ahh.....i was wondering what to do next :0

I was expecting it to happen automatically lol

What is the command to install?

kloknibor, 2016-06-17:
Latest stable:

sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
sudo chmod 777 start.sh
sudo ./start.sh

Latest development:

sudo wget https://github.com/WesGilster/Creation-Workshop-Host/raw/master/host/bin/start.sh
sudo chmod 777 start.sh
sudo ./start.sh WesGilster

currently you started with the stable version so the command will be : 

sudo chmod 777 start.sh
sudo ./start.sh

and it should install

camhguh, 2016-06-17:
ok , installed to this point, looking for a directory that it cant find?

pi@raspberrypi:~ $ sudo chmod 777 start.sh
pi@raspberrypi:~ $ sudo ./start.sh
Installing RxTx
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  librxtx-java
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 167 kB of archives.
After this operation, 848 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main librxtx-java armhf 2.2pre2-13 [167 kB]
Fetched 167 kB in 0s (387 kB/s)
Selecting previously unselected package librxtx-java.
(Reading database ... 31136 files and directories currently installed.)
Preparing to unpack .../librxtx-java_2.2pre2-13_armhf.deb ...
Unpacking librxtx-java (2.2pre2-13) ...
Setting up librxtx-java (2.2pre2-13) ...
Processing triggers for libc-bin (2.19-18+deb8u4) ...
No X server running, starting and configuring one
./start.sh: line 42: startx: command not found
./start.sh: line 43: xhost: command not found
mv: cannot stat ‘cwh-.*.zip’: No such file or directory
Either Java is not installed, or an incorrect version of Java is installed. Installing from this URL: http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz
rm: cannot remove ‘jdk-8u73-linux-arm32-vfp-hflt.tar.gz’: No such file or directory
--2016-06-17 21:05:35--  http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz
Resolving download.oracle.com (download.oracle.com)... 213.104.143.179, 213.104.143.184
Connecting to download.oracle.com (download.oracle.com)|213.104.143.179|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://edelivery.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz [following]
--2016-06-17 21:05:35--  https://edelivery.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz
Resolving edelivery.oracle.com (edelivery.oracle.com)... 104.82.238.98
Connecting to edelivery.oracle.com (edelivery.oracle.com)|104.82.238.98|:443... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://edelivery.oracle.com/osdc-otn/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz [following]
--2016-06-17 21:05:36--  https://edelivery.oracle.com/osdc-otn/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz
Reusing existing connection to edelivery.oracle.com:443.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz?AuthParam=1466197656_d4db0f0580ad10d44f1f3db23293e792 [following]
--2016-06-17 21:05:36--  http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-arm32-vfp-hflt.tar.gz?AuthParam=1466197656_d4db0f0580ad10d44f1f3db23293e792
Connecting to download.oracle.com (download.oracle.com)|213.104.143.179|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 81500626 (78M) [application/x-gzip]
Saving to: ‘jdk-8u73-linux-arm32-vfp-hflt.tar.gz’

jdk-8u73-linux-arm3 100%[=====================>]  77.72M  1.93MB/s   in 42s

2016-06-17 21:06:19 (1.84 MB/s) - ‘jdk-8u73-linux-arm32-vfp-hflt.tar.gz’ saved [81500626/81500626]

Unzipping and installing Java now
Checking for new version from Github Repo: area515/Creation-Workshop-Host
grep: build.number: No such file or directory
Local Tag:
Network Tag: 1.0.0-beta2
Installing latest version of cwh-: 1.0.0-beta2
--2016-06-17 21:06:54--  https://github.com/area515/Creation-Workshop-Host/releases/download/1.0.0-beta2/cwh-1.0.0-beta2.zip
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/27014245/72fe5d62-345d-11e6-8c62-b9e52df19533.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160617%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160617T210655Z&X-Amz-Expires=300&X-Amz-Signature=a7e05cdfdc8e9e4f4a5e25f131247d129630f29fa3b1db39c39712abc4d8a0da&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dcwh-1.0.0-beta2.zip&response-content-type=application%2Foctet-stream [following]
--2016-06-17 21:06:55--  https://github-cloud.s3.amazonaws.com/releases/27014245/72fe5d62-345d-11e6-8c62-b9e52df19533.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160617%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160617T210655Z&X-Amz-Expires=300&X-Amz-Signature=a7e05cdfdc8e9e4f4a5e25f131247d129630f29fa3b1db39c39712abc4d8a0da&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dcwh-1.0.0-beta2.zip&response-content-type=application%2Foctet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.120.235
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.120.235|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33995480 (32M) [application/octet-stream]
Saving to: ‘/tmp/cwh-1.0.0-beta2.zip’

/tmp/cwh-1.0.0-beta 100%[=====================>]  32.42M  1.14MB/s   in 19s
[... 532 lines cut ...]
  inflating: lib/seamless-swing-1.1.0.jar
  inflating: lib/seamless-util-1.1.0.jar
  inflating: lib/seamless-xml-1.1.0.jar
[... cut ...]
  inflating: lib/websocket-common-9.3.7.v20160115.jar
  inflating: lib/websocket-server-9.3.7.v20160115.jar
  inflating: lib/websocket-servlet-9.3.7.v20160115.jar
Turning off screen saver and power saving
./start.sh: line 134: xset: command not found
./start.sh: line 135: xset: command not found
./start.sh: line 136: xset: command not found
Installing CWH as a service
Determinging if one time install has occurred
grep: /root/3dPrinters/config.properties: No such file or directory
Starting printer host server
pi@raspberrypi:~ $

kloknibor, 2016-06-17:
no it's all fine! You just installed photonic3D congratulations!

I bet you now want you use the webinterface, currently your pi is running as a webserver which can communicate with your printer. To acces the web interface there are 2 options for you.

Find with any ip scanner the exact IP of the raspberry pi and add the port :9091 for example : 192.168.1.14:9091 but the first ip will be different most likely.

If you try to acces the webinterface from windows you can also do the following: 

1. download this : https://github.com/area515/Creation-Workshop-Host/releases/download/1.0.0-beta2/cwh-1.0.0-beta2.zip
2. extract
3. run browseprinter.bat if found this will open your printer in the browser!

If you would have used our costum image you would have a third option (so now you don't) which would be to enter photonic3d.local into your browser. This will only work when bonjour is installed on your computer but this ships with things like itunes etc so is installed on quite a few machine already!

Hope this helps. If all runs please close this issue and if you find any feedback or improvements don't hesitate to drop a line here! You can also use our forum on forum.photonic3d.com for any questions or suggestions.

camhguh, 2016-06-17:
OK, Thank you, the Photonic dashboard is displayed correctly!

closed 2016-06-17 by camhguh