area515/Photonic3D #234: [Question] : Which book would you advice

kloknibor, 2016-06-21:
Hi!

I would like to be able to help more and learn more about java. Which books or courses would you guy advice? James mentioned an book about objective programming and how to organize it which I will get soon. But any more advice to learn fast?

Thanks!

Robin

WesGilster, 2016-06-21:
The best way is to learn from experience. Find a small isolated problem that you dislike. Fix it in Java, and submit pull request on github. I'll be critical and give you advise about your fix and that should really help you learn. I'm sure I could open a starter bug that you could work on. Although, we have much more work to do in Javascript than Java.

kloknibor, 2016-06-21:
A starterbug would be fine and javascript is fine too! I'm more familiair with this code but would like to improve in both languages. I must say the problem with the mobile thing I couldn't find so far... Should I retry that again?

WesGilster, 2016-06-21:
No let's try something else much easier. Let's fix that 3d camera angle bug. It's about as easy as it gets and everyone seemed to want it fixed. As you work through this bug look at the code and feel free to ask questions about very strange things. You'll find Angular has some extremely weird ways that it calls functions so let's feel free to ask questions. 

Here is a great Angular tutorial:
http://campus.codeschool.com/courses/shaping-up-with-angular-js/intro

kloknibor, 2016-06-21:
I will start working on it end of the week! Thank you wes! It's appreciated :D! I already did some research to angular and luckily it was well documented! But I will do this tutorial too!

kloknibor, 2016-06-22:
I took a quick glance during my lunch break at the code and found the piece of code which setups the camera's view I suppose : 

//Scene and camera setup
 scene = new THREE.Scene();
camera = new THREE.PerspectiveCamera(45, width / height, 1, 1000 );
camera.position.x = 80;
camera.position.y = -80;
camera.position.z = 300;
camera.up = new THREE.Vector3(0, 0, 1);

I found the documentation of the THREE.PerspectiveCamera over at http://threejs.org/docs/#Reference/Cameras/PerspectiveCamera

kloknibor, 2016-06-22:
ho sorry clicked close and comment by accident! but the function is defined : 

PerspectiveCamera( fov, aspect, near, far )

fov — Camera frustum vertical field of view.
aspect — Camera frustum aspect ratio.
near — Camera frustum near plane.
far — Camera frustum far plane.

And I only understand what they mean with fov. What are the other values? Should I do something with them or should I just calculate the right xyz values for the camera and that's it?

kloknibor, 2016-06-22:
Nevermind I found an good explaination here for who is intrested : https://msdn.microsoft.com/en-us/library/dn479430(v=vs.85).aspx

kloknibor, 2016-06-22:
Hmm I'm having a hard time understanding how to call the different variables. I found what i need which would be : 

var width = printJob.printer.configuration.machineConfig.PlatformXSize;
var length = printJob.printer.configuration.machineConfig.PlatformYSize;
var height = printJob.printer.configuration.machineConfig.PlatformZSize;

But this is just because I found it elsewhere in the code. Could you explain to me how this variables are made together? This call options I mean than. Thanks!

WesGilster, 2016-06-22:
These variables are given to you from the application that asked you to build the 3d component. As an Angular directive, you've been given a variable scope of "printJob" nothing more and nothing less. You don't know why they were sent to you, nor should you care. Your task is to "position" your camera whenever the model changes. Have you found the area of the Angular directive that "watch"es the model as it changes? Where does that lead you?

If you are interested in following the hierarchy of classes that Javascript structure, you start with the first class in Java:

> org.area515.resinprinter.job.PrintJob

eventually you'll make your way to:

> org.area515.resinprinter.printer.MachineConfig

kloknibor, 2016-06-22:
Thanks for your answer wes, I will first do some angularJS courses before bothering you further ;)! You will hear from me!

WesGilster, 2016-06-22:
No problem you aren't bothering. I wasn't telling you to go read a book. I was giving you hints towards how you should find the right piece of code.

"watch"es:
`scope.$watch("printJob", function(newValue, oldValue) {
`
This is how a directive is supposed to know when an internal javascript model changes. Notice how it calls the "loadModel()" method?

"position"
```
camera.position.x = ???;
camera.position.y = ???;
camera.position.z = ???;
```
You've already written the answer to these question marks in your statement above.

"printJob"
```
scope: {
   printJob: "=printJob"
},
```

This is how external callers give model information into the directive.

WesGilster, 2016-06-23:
Let's continue this conversation in bug #228. It's conversation is much more relevant there.

closed 2016-06-23 by WesGilster