area515/Photonic3D #235: Allow user to specify default repo to update from

jmkao, 2016-06-23:
The stock default repo is area515/Creation-Workshop-Host. In the old `start.sh`, the update logic was based on updating when the remote version was bigger than the current version. Thus if you switched to a dev repo once with a higher number, that build would stick for awhile, at least until the stable repo advanced to that build number.

Now, the logic is based on picking whatever the current release of the specified repo is, with no notion of a monotonically increasing and consistent concept of "newer". This may make more sense with multiple forks, but means that if you are using a dev repo, you always need to specify the repo as a parameter to `start.sh`, otherwise you will get reverted to the stable repo. This also means that whenever the box is rebooted, the install will be reverted to the stable repo unless the user makes modifications to /etc/init.d/cwhservice.

It should be possible for a user to persistently specify the repo they want their install to use. It is possible to expand the use of ~/3dPrinters/config.properties to look for an `updateRepo` property. If the property is present, then its value should be used as the update repo when one is not specified at the command line.

WesGilster, 2016-06-23:
This should automatically happen. Any property in ~/3dPrinters/config.properties should automatically override the property in the regular config.properties. That's how you are able to "save" properties without getting them reverted when you reinstall Photonic. Maybe I don't quite understand what you are asking for though.

jmkao, 2016-06-23:
This is a bug to myself mainly :).

I need to code start.sh to look at config.properties since it runs outside of Java.

WesGilster, 2016-06-23:
Ahh. Nevermind then...

closed 2016-06-27 by WesGilster