area515/Photonic3D #236: [Bug] : Dev version only shows topbar

kloknibor, 2016-07-03:
Hi,

I was just trying to test the new dev version but I only get the topbar in the browser. I tried this on windows 7 (running and browsing from). first it showed the dashboard but it disappeared and even after restarting the software the problem persists. Here is the page source : 

<!DOCTYPE html>

<html lang="en" ng-app="cwhApp">
 <head>
  <base href="/">
  <meta http-equiv="Content-Type" content="text/html; charset=UTF-8" >
  <meta charset="utf-8">
  <meta http-equiv="X-UA-Compatible" content="IE=edge"> 
  

    <meta name="mobile-web-app-capable" content="yes">
    <meta name="viewport" content="width=device-width, initial-scale=1.0, maximum-scale=1.0, minimum-scale=1.0, user-scalable=no">
	  
  <link rel="apple-touch-icon" sizes="57x57" href="/favicon/apple-icon-57x57.png">
<link rel="apple-touch-icon" sizes="60x60" href="/favicon/apple-icon-60x60.png">
<link rel="apple-touch-icon" sizes="72x72" href="/favicon/apple-icon-72x72.png">
<link rel="apple-touch-icon" sizes="76x76" href="/favicon/apple-icon-76x76.png">
<link rel="apple-touch-icon" sizes="114x114" href="/favicon/apple-icon-114x114.png">
<link rel="apple-touch-icon" sizes="120x120" href="/favicon/apple-icon-120x120.png">
<link rel="apple-touch-icon" sizes="144x144" href="/favicon/apple-icon-144x144.png">
<link rel="apple-touch-icon" sizes="152x152" href="/favicon/apple-icon-152x152.png">
<link rel="apple-touch-icon" sizes="180x180" href="/favicon/apple-icon-180x180.png">
<link rel="icon" type="image/png" sizes="192x192"  href="/favicon/android-icon-192x192.png">
<link rel="icon" type="image/png" sizes="32x32" href="/favicon/favicon-32x32.png">
<link rel="icon" type="image/png" sizes="96x96" href="/favicon/favicon-96x96.png">
<link rel="icon" type="image/png" sizes="16x16" href="/favicon/favicon-16x16.png">
<link rel="manifest" href="/favicon/manifest.json">
<meta name="msapplication-TileColor" content="#ffffff">
<meta name="msapplication-TileImage" content="/favicon/ms-icon-144x144.png">
<meta name="theme-color" content="#ffffff">

  	<title>Photonic3D</title>
  	
	
		<link href="./bootstrap/css/bootstrap.min.css" rel="stylesheet">
		<link href="./bootcards/css/bootcards-desktop.min.css" rel="stylesheet" type="text/css">
		<link href="./fontawesome/css/font-awesome.min.css" rel="stylesheet">
		<link href="./cwh/css/cwh.css" rel="stylesheet">
		<link href="./chart/angular-chart.min.css" rel="stylesheet">
	
	
	
	<script>var isDesktop = true;</script>
	
	
    

    <script src="./jquery/jquery.min.js"></script>
	<script src="./bootstrap/js/bootstrap.min.js"></script>
	<script src="./jquery/jquery-ui.min.js"></script>
	<script src="./bootcards/js/bootcards.min.js"></script>
	<script src="./angular/js/angular.min.js"></script> 
	<script src="./angular/js/angular-route.min.js"></script> 
	<script src="./angular/js/angular-animate.js"></script>
	<script src="./chart/Chart.min.js"></script>
	<script src="./chart/angular-chart.min.js"></script>
	<script src="./angularui/ui-bootstrap-tpls-1.1.2.min.js"></script>
	
	<script src="./cwh/js/cwhspinner.js"></script>
	<script src="./cwh/js/cwhtestscript.js"></script>
	<script src="./cwh/js/cwhsocket.js"></script>
	<script src="./cwh/js/cwhcomport.js"></script>
	<script src="./cwh/js/index.js"></script> 
	<script src="./cwh/js/dashboard.js"></script> 
	<script src="./cwh/js/printers.js"></script> 
	<script src="./cwh/js/printerControls.js"></script> 
[... 155 lines cut ...]
	</div>

	
	<div class="modal fade" id="docsModal" tabindex="-1" role="dialog" aria-labelledby="docsModal" aria-hidden="true">
	  <div class="modal-dialog">
	    <div class="modal-content"></div>
	  </div>
	</div>
	<div class="modal fade" id="errorModal" tabindex="-1" role="dialog" aria-labelledby="errorLabel" aria-hidden="false" >
	  <div class="modal-dialog">
	    <div class="modal-content">
	      <div class="modal-header">
	        <button type="button" class="close" data-dismiss="modal" aria-label="Close"><span aria-hidden="true">&times;</span></button>
	        <h4 class="modal-title" id="errorModalTitle">{{currentError.command}}</h4>
	      </div>
	      <div class="modal-body" id="errorModalBody">
	      {{currentError.message}}
	      </div>
	      <div class="modal-footer">
	        <button type="button" class="btn btn-default" data-dismiss="modal">Close</button>
	      </div>
	    </div>
	  </div>
	</div>
</body></html>

WesGilster, 2016-07-03:
Are you testing with SSL turned on?

WesGilster, 2016-07-03:
I'm going with the assumption you are using SSL. I just tested this on Edge and Chrome and SSL and login seems to work. If you haven't added any users through the migrate feature, you are going to be effectively locked out of the application.

kloknibor, 2016-07-03:
I haven't done anything in the config yet. So I suppose it should be turned off? I will test later, I've became sick today.

Verstuurd vanaf mijn iPhone

kloknibor, 2016-07-14:
This was linked to the other problems, the topbar only showing was just google chrome showing it's cash of photonic3d

closed 2016-07-14 by kloknibor