area515/Photonic3D #237: [Set-up problems]: photocentric LC10 image

kloknibor, 2016-07-07:
Hi,

I had somebody over at the issues of my photocentric fork of photonic3D. But I think it is best to move the conversation to here to keep everything centralized! So the issues were : 

1. Z axis don't move during the printing.
2. Resolution was set to 1024x600 into config file but Sliced STL was not the good measurement
3. Image did not run properly (don't worry it was my image not James)

log bundles : 
https://github.com/kloknibor/photocentric_dev/files/345234/LogBundle.zip [ just turning on, not much usefull information]
https://github.com/kloknibor/photocentric_dev/files/348375/LogBundle.zip [during printing]

Don't worry I'm on this issue! I just screwed up the templates and should change them to our freemaker ones. Also the dots per mm are most likely wrong and I'll look into the image too. 

Please let me know if you don't like this conversation here but rather back at my repo or even on photonic3d.com and I'll move it again ;)! But I think it is good for the moral too hear about the active users :)!

I agreed to him/her to look and test this coming weekend ;)!

Robin

WesGilster, 2016-07-07:
I've mentioned this before on another thread, but your gcode templates are bad in the photocentric printer you are advertising on github. Look at what you are sending to the printer:
2016-07-05 17:49:36,140 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] Write 0: M650 D9 S0,25

2016-07-05 17:49:36,442 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] lineRead: error: parse failure: can't parse M code at line 'M650 D9 S0,25'

2016-07-05 17:49:36,455 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-1] Write 0: G1{$SlideTiltVal != 0? X$SlideTiltVal:} Z(9 * 1) F{0 < 4?$ZBottomLiftRate:0,25}

WesGilster, 2016-07-07:
What you should see in the logs are what the printer should take.

kloknibor, 2016-07-07:
I saw indeed, I still need to change it to freemarker templates like I mentioned above so indeed my bad ;)! On it!

WesGilster, 2016-07-08:
Oh cool and to answer your other question. I think those problems are relevant to ask here instead of another github fork as long as you are interested in me and others in the community to help answer them. Eventually this may become a developer only forum and the Photonic forum will answer user questions, but I see a mixture until that split becomes more apparent to the community.

kloknibor, 2016-07-10:
So I'm working on the templates now but I'm wondering if I can add an delay in the template? I noticed that @jmkao uses G4 S2 to let the resin settle. But since the firmware of photocentric is really primitive (and I lost it currently so can't check) I was wondering if I could add the delay in another way?

Thanks!

kloknibor, 2016-07-10:
I can confirm that indeed this code isn't available in my firmware (remembered somewhere were I saved it :))

closed 2016-07-10 by kloknibor