area515/Photonic3D #239: [Question]: Image creation fails

kloknibor, 2016-07-10:
Hi,

I'm trying to maintain an photocentric image for my printer and it does build fine and I upload it to the server without problems now. But when I try to make the image from my repo or from area515 repo it does build but when started only the main page without any /$currentpage_like_dashboard_here works but the only tab that shows is dashboard and when refreshing it crashes. 

The build process can be found here : https://semaphoreci.com/kloknibor/photocentric_build_photonic3d and should be the same as the one @jmkao has... So I'm in the black what is wrong since it does something but doesn't install cwh properly!

Could you point me in the right direction to fix this?

Thanks!

Robin

jmkao, 2016-07-10:
According to the log in build 63 in semaphore, you are still installing from area515.

kloknibor, 2016-07-11:
Yes I changed that back so everything should be the same but even than only dashboard shows up. I know I need to change the photonic3D start url to use my repo for the image creation. But as long as I don't even get area515 installing the right way I won't be able to do my own repo either was what I thought ;)! The builds before the last one were all with my own repo.

kloknibor, 2016-07-12:
@jmkao I tried some more, all again espacially with the area515 one because this has been build without a problem at some point. And even that still fails.

The only thing I changed from the area515 OctoPi in this repo : https://semaphoreci.com/kloknibor/octopi/ is that repo url. I changed WesGilster to area515 (Because the Wesgilster one didn't want to build)

Do you see anything I set up wrong?

jmkao, 2016-07-12:
If the image builds successfully, then the question would  be why the Photonic install is inoperable, since start.sh is pretty much the same no matter where it's used.

When you go into the RPi image after you've installed it, can you identify some problem with the install in /opt/cwh?

If you can figure out what is wrong in the image, we can backtrack to see what is different about the install process.

kloknibor, 2016-07-13:
Sorry I haven't been at home yet to check the installed version. I have however compared the uninstalled images files (using OSFmount and exfat32). 

in the normal readable partition those file are different (Probably irrelavant):
[image]
[image]

But in the linux partition there are different files... In your script /opt/ only conatains a folder "vc" in mine there is also the file "cwh" : 

[cwh.pdf](https://github.com/area515/Creation-Workshop-Host/files/361173/cwh.pdf)

(remove .pdf ... on this pc zip uploading still doesn't work)

and in the root these 2 files are present in my version : 

[common.sh.txt](https://github.com/area515/Creation-Workshop-Host/files/361185/common.sh.txt)
[start.sh.txt](https://github.com/area515/Creation-Workshop-Host/files/361186/start.sh.txt)

While in yours this are the available files : 

[chroot_script.txt](https://github.com/area515/Creation-Workshop-Host/files/361187/chroot_script.txt)
[common.sh.txt](https://github.com/area515/Creation-Workshop-Host/files/361188/common.sh.txt)

(added the .txt also manually for uploading here)

It's weird we got different files while using the same commands etc... Does this tell you more?

jmkao, 2016-07-13:
No idea what that means. Focus on what makes Photonic3D run differently or incorrectly in your image.

The question is, what do you need to do to fix the Photonic3D installation on your image, then whatever those steps are, we can use that to figure out how to generate a working image.

kloknibor, 2016-07-28:
Sorry for the late replay, been so busy... But I did got some time to look into it at my lunch break at work haha. (have got a pi here to for testing work related stuff)

Lets correct myself first, Photonic3d does NOT start. What I was seeing were cached browser files sorry.

On boot I get the following error : 

> [FAILED} Failed to start LSB: Starts the CWH.
> see 'systemct1 statys cwhservice.service' for detail

In this file there is the following code : 

> # Automatically generated by systemd-sysv-generator
> 
> [Unit]
> SourcePath=/etc/init.d/cwhservice
> Description=LSB: Starts the CWH
> Before=runlevel2.target runlevel3.target runlevel4.target runlevel5.target shutdown.target
> After=all.target
> Conflicts=shutdown.target
> 
> [Service]
> Type=forking
> Restart=no
> TimeoutSec=5min
> IgnoreSIGPIPE=no
> KillMode=process
> GuessMainPID=no
> RemainAfterExit=yes
> SysVStartPriority=4
> ExecStart=/etc/init.d/cwhservice start
> ExecStop=/etc/init.d/cwhservice stop
> 

I know the services have to with making it able to start and stop photonic3d as service but really that is were my knowledge ends... 

Any help would be appriciated :) 

Thanks

Robin

WesGilster, 2016-07-28:
Correct me if I'm wrong @jmkao. Ultimately your entire image should be identical to our stock image except for the following line:

> ./start.sh

That should something like:

> ./start.sh nameOfYourRepo

Keep in mind that there is currently a bug that reverts everyone's repos back to the original until this bug is fixed: https://github.com/area515/Creation-Workshop-Host/issues/242

Also, you should know, this is the easiest part of maintaining your own fork as it's just a single line script change. Keeping your users updated with the latest changes merged into your fork is ongoing maintenance concern. Your fork is already quite a bit behind. If there are specific changes that you'd like put into the GUI, there are MUCH easier ways that you can get the GUI customized/branded and keep service side compatibility. Just trying to make your life much easier!

jmkao, 2016-07-28:
Actually, the way to make the default repo a different fork would be to set /root/3dPrinters/config.properties as described in #242 before running start.sh the first time.

That should install from the fork, and will make the default fork for a plain `start.sh` call to be whatever repo is specified there.

kloknibor, 2016-07-29:
@WesGilster @jmkao Currently the only thing different are 2 settings in config.properties and added new GUI resources.

first thing is like @jmkao stated to set the repo value to my repo to pull only photocentric printers etc.

Second thing is set the resources to the correct one. I can just do a merge and manually update the GUI resources, but these change are minor if I keep it up to date :)

I thought this was already quite easy but any suggestions are welcome! I will do a merge now ;)! Haven't been so active lately, sorry.

kloknibor, 2016-07-29:
Oh and I've been an idiot, instead of looking into the file I should have run : `systemct1 statys cwhservice.service` to see the error codes, will do that later.

kloknibor, 2016-08-03:
This command won't run... (Isn;t recognized) is it something that has been deleted? When running sudo apt-get install systemct1 it won't find the needed packages

kloknibor, 2016-08-03:
This is the error log : 

```

● cwhservice.service - LSB: Starts the CWH
   Loaded: loaded (/etc/init.d/cwhservice)
   Active: failed (Result: exit-code) since Fri 2016-05-27 11:42:09 UTC; 2 months 7 days ago
  Process: 417 ExecStart=/etc/init.d/cwhservice start (code=exited, status=203/EXEC)
```

kloknibor, 2016-08-03:
The script in init.d is identical and the start.sh seems to be fine... Also the software is unpacked in opt/cwh and totally fine...! I can manually install photonic3D on top of this image and it will work fine, but I'm not sure what steps I should take to fix it at this moment. The cwhservice is the same too by the way.

Could anybody point me in the right directions?

WesGilster, 2016-08-05:
Can I get a precise diff between your image and the stock image? In an above post it looked like you mentioned there were several files that were different. You can't have that. 

From what James said, there should be one line difference in your startup:
`cd ~; printf 'updateRepo=[YourRepo]/Creation-Workshop-Host\n' > config.properties`

That will create the file you are looking for so start.sh will start properly

kloknibor, 2016-08-05:
Indeed some files are different, yesterday It seemed like I had 1 succesfull build of area515 and switched after that to my own branch but no luck... Can't replicate the succesfull one either.

Also yesterday an start.sh appeared in the root after running yhe image with my repo and it was corrupt or something, it was computer language instead of plain text.

Could it be possible that github stops me from downloading the start.sh? Like we had before with the images?

I will upload the contents off both files later this week. Also I tried finding the string which defines the repo url in the image but couldn't find it :(!

jmkao, 2016-08-05:
@kloknibor  My suggestion would be to first work on getting a functional build that works reliably that is an exact copy of what I am doing. Then from there, you can experiment with adding additional functionality to create the necessary config.properties. At this point, I really have no idea what changes you have made to any of the scripts, so it's hard to say why there is a difference in behavior.

kloknibor, 2016-08-05:
Thanks both for chipping in. @jmkao That's exactly what I did before. I will make a new fork with an exact clone of your octopi image so it will use are515 resources. This is what I was already doing but maybe I changed things I forgot about. The only thing I do different is that I use FTP to upload to the server since I don't have an s3 cloud... Will do this tonight or in the weekend, will keep you updated ;)

kloknibor, 2016-08-05:
I have made a exact copy of what james did here : https://photonic3d.com/downloads/Photocentric_photonic3d_test.zip I haven't got time to test it yet, and have to get soon up for work already, I'll test it tomorrow. May I ask you @jmkao if you could make a image of photonic3d from area515 today? This way I can easily compare them and make sure the differences are photonic3d related and not just updated tools in the OS related.

Thanks in advance!

Robin

jmkao, 2016-08-06:
I tested this using a custom default repo (populating /root/3dPrinters/config.properties) in OctoPi and making a fork-specific image for Kudo3D's fork, and it pretty much worked.

Remember that customizations that you make to the image creation process need to go inside the post_chroot_script for the photonic3d, and you need to pull down the start.sh that is in Wes or my repo, not the one in area515, which doesn't have the ability to configure the default update repo.

https://github.com/Kudo3D/OctoPi
https://semaphoreci.com/jkao/kudo3d-octopi

kloknibor, 2016-08-07:
Ho james,

I did try it with the area515 one, that might be the issue than! Tomorrow I have madr some hours to work on this. I will remake the image from wesgilster repo or let me know the location of uour repo so I can reproduce your work (or is it just over at octipi @jkao?)  currently it's 4:39 in the morning here and just coming from the bar, so first some sleep :)! Hope to get it working tomorrow :)!

How is the work from the interns coming slong by the way? Becominging something sweet? I would love to hear the progress!

Robinsweet dreams for now :)

kloknibor, 2016-08-07:
I can make images from the WesGilster repo just fine at the moment :)! So got a good starting point, thanks for sticking with me 

Robin

closed 2016-08-07 by kloknibor

kloknibor, 2016-08-07:
Oh one of the changes I did is using the deploy branch where I otherwise used the devel branch. Maybe this caused issues... Will test that out tomorrow.

kloknibor, 2016-08-07:
And I was testing with area515 with the philosophy that that should work because there was an image already haha ;)!