area515/Photonic3D #240: NegativeArraySizeException loading STL File

jmkao, 2016-07-11:
The attached STL file is an ASCII format STL file that begins with the string `solid OpenSCAD_Model` and was generated by OpenSCAD.

Looking at the stack trace, it would seem that the file is being loaded as a binary STL rather than an ASCII STL file.

```
Caused by: java.lang.NegativeArraySizeException
        at org.area515.resinprinter.slice.StlFile.readBinaryFile(StlFile.java:317) ~[main/:?]
        at org.area515.resinprinter.slice.StlFile.load(StlFile.java:433) ~[main/:?]
        at org.area515.resinprinter.slice.ZSlicer.loadFile(ZSlicer.java:939) ~[main/:?] 
        at org.area515.resinprinter.job.STLFileProcessor.processFile(STLFileProcessor.java:88) ~[main/:?]
        at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:38) ~[main/:?]
        at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:11) ~[main/:?]
```

I suspect that the logic in in commit f1b20e84cb40defe84a2f3ab3e7c764b87183222 for #226 is having trouble here, but I'm not sure where the `facet` seeking code is going wrong.

[lenscap-36mm.stl.zip](https://github.com/area515/Creation-Workshop-Host/files/356265/lenscap-36mm.stl.zip)

WesGilster, 2016-07-11:
Yes it is. I'll check it out and add a test...

WesGilster, 2016-07-11:
Fixed and checked in a test for it.

jmkao, 2016-07-11:
Cool. Did you push this fix to your repo yet?

WesGilster, 2016-07-11:
Yep, but no build. I figure everyone can do that now.

jmkao, 2016-07-11:
Hmm, strange, I'm not seeing any commits after the merge of PR7 on WesGilster/Creation-Workshop-Host

WesGilster, 2016-07-11:
I pushed and didn't check the result, it failed. Pulled down Robin's changes and repushed.

jmkao, 2016-07-11:
Awesome, confirmed this works. Thanks!

closed 2016-07-11 by jmkao