area515/Photonic3D #244: [Bug] : Github denies acces after some refreshes on printer choosing screen

kloknibor, 2016-08-03:
Hi,

I was having trouble getting my profile up and running, so I refresed the "choose a printer"  page sometimes maybe like 10 times max? After that when trying to get the resources you will get the following error from the chrome console : 

[image]

I think we should start by putting the resources on the server... Do you guys agree?

kloknibor, 2016-08-03:
After waiting for a while (15 minutes maybe sooner) it worked again. But still I think it wouldn't harm to move such resources to our own server. But I was thinking how we could do this efficiently. I can just make a folder with the needed files and use the defined repo from the config file to identify the needed folder. For this way I need to create manually the output from the github api. THe output I'm talking about is this : 

```
[
  {
    "name": "Titan1.json",
    "path": "host/printers/Titan1.json",
    "sha": "496b3d10cef11e3f937f5e6126fc44ab8b538f5b",
    "size": 4673,
    "url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/Titan1.json?ref=master",
    "html_url": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/Titan1.json",
    "git_url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/496b3d10cef11e3f937f5e6126fc44ab8b538f5b",
    "download_url": "https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/master/host/printers/Titan1.json",
    "type": "file",
    "_links": {
      "self": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/Titan1.json?ref=master",
      "git": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/496b3d10cef11e3f937f5e6126fc44ab8b538f5b",
      "html": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/Titan1.json"
    }
  },
  {
    "name": "mUVe 1.json",
    "path": "host/printers/mUVe 1.json",
    "sha": "944a6fbeb7192c1564950b0ea6d6aab029327cd7",
    "size": 4498,
    "url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/mUVe%201.json?ref=master",
    "html_url": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/mUVe%201.json",
    "git_url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/944a6fbeb7192c1564950b0ea6d6aab029327cd7",
    "download_url": "https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/master/host/printers/mUVe%201.json",
    "type": "file",
    "_links": {
      "self": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/mUVe%201.json?ref=master",
      "git": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/944a6fbeb7192c1564950b0ea6d6aab029327cd7",
      "html": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/mUVe%201.json"
    }
  },
  {
    "name": "photocentric 10.json",
    "path": "host/printers/photocentric 10.json",
    "sha": "28f77c09645b98401d8410ef943175c773070973",
    "size": 3538,
    "url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/photocentric%2010.json?ref=master",
    "html_url": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/photocentric%2010.json",
    "git_url": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/28f77c09645b98401d8410ef943175c773070973",
    "download_url": "https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/master/host/printers/photocentric%2010.json",
    "type": "file",
    "_links": {
      "self": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/contents/host/printers/photocentric%2010.json?ref=master",
      "git": "https://api.github.com/repos/WesGilster/Creation-Workshop-Host/git/blobs/28f77c09645b98401d8410ef943175c773070973",
      "html": "https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/photocentric%2010.json"
    }
  }
]
```

Thats not really efficient and I think eventually we want that the community could submit new profiles won't we? 

Any idea how we can do this better? If not I can make the proposed implementation tomorrow.

Robin

jmkao, 2016-08-03:
If there is a rate limit in place, we could include the file as part of a release, which will locate the file in what is likely to be a more accessible location. I would, as much a possible, prefer to keep any network dependencies only on GitHub rather than our own servers.

kloknibor, 2016-08-03:
Hmm I do like that, It seems to be an rate limit in the api calls from github (the content function) so indeed we could easily add that in the release commands, Good idea!

WesGilster, 2016-10-15:
It seems you guys already fixed this with the recent commit. It seems like the problem all along was that Githug doesn't forward requests through renamed repos. If you guys feel differently, let me know, otherwise I'll close.

WesGilster, 2016-10-19:
Fixed now.

closed 2016-10-19 by WesGilster