area515/Photonic3D #247: Can't build anymore

kloknibor, 2016-08-08:
Hi,

I'm having trouble building after the latest pulls from Wesgilster and even area515. I haven't changed a thing in my set-up only ran windows update and installed a game and the commands java and javac both run fine.

Does it still run fine with you guys? If so it will be something on my local machine. But I'm unsure what it would be since javac an java run fine, java sdk is installed. The command` gradlew` runs fine. But `gradlew run `and `gradlew fullRelease` fail

This is the debug log : 

11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.dependencies using tasks.addPlaceholderAction(dependen
cies)
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.dependencies' to state Discovered.

11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.dependencyInsight' rule action tasks.add
PlaceholderAction(dependencyInsight)
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.dependencyInsight using tasks.addPlaceholderAction(dep
endencyInsight)
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.dependencyInsight' to state Discov
ered.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.distTar' rule action Project.<init>.task
s.distTar()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.distTar using Project.<init>.tasks.distTar()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.distTar' to state Discovered.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.distZip' rule action Project.<init>.task
s.distZip()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.distZip using Project.<init>.tasks.distZip()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.distZip' to state Discovered.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.eclipse' rule action Project.<init>.task
s.eclipse()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.eclipse using Project.<init>.tasks.eclipse()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.eclipse' to state Discovered.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.eclipseClasspath' rule action Project.<i
nit>.tasks.eclipseClasspath()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.eclipseClasspath using Project.<init>.tasks.eclipseCla
sspath()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.eclipseClasspath' to state Discove
red.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.eclipseJdt' rule action Project.<init>.t
asks.eclipseJdt()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.eclipseJdt using Project.<init>.tasks.eclipseJdt()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.eclipseJdt' to state Discovered.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Running model element 'tasks.eclipseProject' rule action Project.<ini
t>.tasks.eclipseProject()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Mutating tasks.eclipseProject using Project.<init>.tasks.eclipseProje
ct()
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
roject : - Transitioning model element 'tasks.eclipseProject' to state Discovere
d.
11:09:15.781 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] P
[... 213 lines cut ...]
11:09:16.431 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine
.store.CachedStoreFactory] Resolution result cache closed. Cache reads: 0, disk
reads: 0 (avg: 0.0 secs, total: 0.0 secs)
11:09:16.431 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine
.store.CachedStoreFactory] Resolution result cache closed. Cache reads: 0, disk
reads: 0 (avg: 0.0 secs, total: 0.0 secs)
11:09:16.431 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine
.store.ResolutionResultsStoreFactory] Deleted 2 resolution results binary files
in 0.0 secs
11:09:16.431 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.me
mcache.InMemoryCachedRepositoryFactory] In-memory dependency metadata cache clos
ed. Repos cached: 3, cache instances: 3, modules served from cache: 0, artifacts
: 0
11:09:16.431 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCach
e] Closing cache plugin-use-metadata.bin (C:\Users\Robin\.gradle\caches\2.14\plu
gin-resolution\plugin-use-metadata.bin)
11:09:16.431 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasin
g lock on Plugin Resolution Cache (C:\Users\Robin\.gradle\caches\2.14\plugin-res
olution).
11:09:16.441 [DEBUG] [org.gradle.api.internal.tasks.compile.daemon.CompilerDaemo
nManager] Stopping 0 compiler daemon(s).
11:09:16.441 [INFO] [org.gradle.api.internal.tasks.compile.daemon.CompilerDaemon
Manager] Stopped 0 compiler daemon(s).

C:\Users\Robin\Desktop\photocentric_dev\host>

WesGilster, 2016-08-08:
I'll try it out now.

WesGilster, 2016-08-08:
Still works fine, I just built DEV016. Make sure you setup your gradle.properties if this is a new setup.

jmkao, 2016-08-08:
The debug log doesn't really help, since that's more for debugging gradle than for debugging a build. What happens when you run it normally?

kloknibor, 2016-08-08:
I used a new fork from WesGilster but all other forks fail too. I checked my gradle.properties and that only contains the release repo and the github key. Which are both fine and it exits on compilejava so it shouldn't have to do something with github yet I suppose. Als gradlew run fails.

kloknibor, 2016-08-08:
@jmkao what do you mean with run it normally? The software runs when I download a release from github and with the normal gradlew command it runs and says build succefull.

WesGilster, 2016-08-08:
He's saying that the debug statements are getting in the way of the actual problem which is:
No configured remotes match the the remote named: upstream

Which means your gradle.properties is probably bad:
`throw new GradleException("No configured remotes match the the remote named: ${releaseRepoRemoteName}");`

kloknibor, 2016-08-09:
@WesGilster You are right, this was part of the problem but it still won't build here is the normal gradle run : 

```
D:\Github\Photonic3D_dev\host>gradlew run
:compileJava
D:\Github\Photonic3D_dev\host\src\main\java\org\area515\resinprinter\security\Je
ttySecurityUtils.java:40: error: cannot find symbol
import org.area515.resinprinter.util.security.LdapUtils;
                                             ^
  symbol:   class LdapUtils
  location: package org.area515.resinprinter.util.security
D:\Github\Photonic3D_dev\host\src\main\java\org\area515\resinprinter\security\ke
ystore\KeystoreUtilities.java:23: error: cannot find symbol
import org.area515.resinprinter.util.security.LdapUtils;
                                             ^
  symbol:   class LdapUtils
  location: package org.area515.resinprinter.util.security
2 errors
:compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Compilation failed with exit code 1; see the compiler error output for details
.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug
option to get more log output.

BUILD FAILED

Total time: 9.417 secs

```

kloknibor, 2016-08-09:
This can be closed. I tried a new clone from your repo @WesGilster and it seems to work now again :D Thanks for the help!

closed 2016-08-09 by kloknibor

WesGilster, 2016-08-09:
That second problem is a missing file due to a bad check-in to git-hub that I made.