area515/Photonic3D #248: Test Failure, NPE from org.area515.resinprinter.security.KeystoreSecurityTest.attemptConversationWithoutAKeyExchange

jmkao, 2016-08-12:
From WesGilster repo.

```
org.area515.resinprinter.test.FullTestSuite > org.area515.resinprinter.security.KeystoreSecurityTest.attemptConversationWithoutAKeyExchange FAILED
    java.lang.NullPointerException
        at org.area515.resinprinter.util.security.PhotonicCrypto.buildEncryptedMessage(PhotonicCrypto.java:336)
        at org.area515.resinprinter.security.KeystoreSecurityTest.attemptConversationWithoutAKeyExchange(KeystoreSecurityTest.java:217)
```

Specifically, PhotonicCrypto.java:336 is:

```
		String[] userIdAndName = LdapUtils.getUserIdAndName(remoteCrypto.verifier.getSubjectDN().getName());
```

The only things that look like they could be null are `remoteCrypto` or `remoteCrypto.verifier`.

WesGilster, 2016-08-12:
Fixed and checked in, but you'll need to use this as a JVM option for RendezvousServer:
-noverify

I'll fix the jvm option in gradle as I don't have time now.

jmkao, 2016-08-12:
Looks good, tests pass now. Thanks!

WesGilster, 2016-08-13:
Weird. Strange that they are passing without the -noverify. It must be a Java version thing? Just keep this thread in mind if you see this:
```
java.lang.VerifyError: Bad <init> method call from inside of a branch
Exception Details:
  Location:
    org/area515/resinprinter/security/keystore/RendezvousExchange$HttpRequestRunnable.<init>(Lorg/area515/resinprinter/security/keystore/RendezvousClient;Ljava/util/UUID;Ljava/util/UUID;Lorg/area515/resinprinter/security/keystore/RendezvousExchange$HttpRequestRunnable;)V @56: invokespecial
  Reason:
    Error exists in the bytecode
  Bytecode:
    0x0000000: 2a2b 2c2d 3a08 3a07 3a06 3a05 12a2 b800
    0x0000010: 9a06 bd00 0459 0319 0653 5904 1907 5359
    0x0000020: 0519 0853 1301 0eb8 0030 b801 113a 0a19
    0x0000030: 0ab2 003a a500 0a2a b700 1ea7 0010 1905
    0x0000040: 1906 1907 1908 b701 1301 57b1          

```

closed 2016-08-13 by WesGilster

jmkao, 2016-08-14:
Yes, it seems like the problem is fixed in 1.8.0_25 and higher:

https://issues.apache.org/jira/browse/GROOVY-6951