area515/Photonic3D #249: build in dev repo not working again

kloknibor, 2016-08-14:
Hi there seems to be an small bug in the dev repo currently, this is the gradle log when trying to release it : 
```

D:\Github\Creation-Workshop-Host\host>gradlew fullRelease
:setupRemote
Releasing to: [https://github.com/Photocentric3D/Creation-Workshop-Host.git]
Repo Owner: Photocentric3D / Repo Name: Creation-Workshop-Host
:setupRelease
----------------------------------------------
Description of current commit: DEV016-11-g2c71284
UNCOMMITTED STAGED CHANGES: org.ajoberstar.grgit.Status$Changes(added:[], modifi
ed:[], removed:[])
UNCOMMITTED UNSTAGED CHANGES: org.ajoberstar.grgit.Status$Changes(added:[], modi
fied:[], removed:[])
----------------------------------------------
Dry-Run Git Push to origin:
----------------------------------------------
Check the above git push dry run to ensure that your push will succeed!
[ant:input] Press <return> to continue...
> Building 5% > :setupRelease
Recommended development release tag: Photocentric3D-2c71284
[ant:input] Enter the tag to create for this release >
> Building 5% > :setupReleaseChange-Repo
Added tag: Change-Repo
:compileJava
Note: D:\Github\Creation-Workshop-Host\host\src\main\java\org\area515\resinprint
er\security\keystore\ByteSessionInputBuffer.java uses or overrides a deprecated
API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processResources
:classes
:jar
:startScripts
:compileTestJava
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processTestResources
:testClasses
:testJar
:installDist
:repoVersion
Git Commit Description: Change-Repo
:preDist
:BuildCWH
:dist
:releaseDist
Releasing the following files: [D:\Github\Creation-Workshop-Host\host\cwh-Change
-Repo.zip]
:githubRelease FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':githubRelease'.
> org.apache.http.client.ClientProtocolException

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug
option to get more log output.

BUILD FAILED

Total time: 2 mins 0.284 secs

```
So the cause is 

> org.apache.http.client.ClientProtocolException

WesGilster, 2016-08-15:
All tests pass and I just built DEV-17.

WesGilster, 2016-09-18:
Not reproducible.

closed 2016-09-18 by WesGilster