area515/Photonic3D #252: Change root password of the PI with our image creation

kloknibor, 2016-09-11:
Hi!

Sorry for being a bit inactive lately! I haven't forgotten your guys but I'm busy preparing for moving and finishing up in the Netherlands ;)! 

First I would like to notify you that we are moving to a new server which is the fastest I tested (quick international speeds, better cpu(also 2 instead of one) and we will have faster than SSD disks :)! I will move Photonic3D in the coming days ;)! 

Secondly I was a bit astonished to find quite a lot of brute force attacks on the new server (Since I didn't set up the firewall and blocker yet) on the user "pi" this means hackers are trying to get acces to internet connected pi's and maybe use them for other attacks without the user knowing. Since all the passwords for SSH on the pi are the same this will be quite simple.

I know most users won't connect their pi yet to the internet (I might be the only idiot at this point :P) but would it be a good idea to change the password of the user pi by default? This could be easily done in the image creation scripts with only a passwd command. What would your opinion on this be?

Thanks in advance!

Robin

WesGilster, 2016-09-11:
Yeah, nothing wrong with that. If users install with the browseprinter 
scripts, they are already prompted to change the ssh password on the pi. 
Otherwise I can ask the user if they want to sync the ssh password with 
the user password when the user provisioning release is complete.

Thanks,

Wes G.

WesGilster, 2016-11-11:
I assume that this should be closed? Please open if you'd like more discussion on the topic.

closed 2016-11-11 by WesGilster