area515/Photonic3D #260: Odroid install not working

kloknibor, 2016-09-30:
Hi guys,

I'm trying different hardware to run photonic3d on, one of them is an Odroid C2. The first problem was that Curl was missing so I installed that but it still doesn't start the server. It also give some errors about the screen which isn't able to select.

Here is the log :  

`root@odroid64:~# sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh

--2016-09-30 10:10:19--  https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
Resolving github.com (github.com)... 192.30.253.113
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://github.com/area515/Photonic3D/raw/master/host/bin/start.sh [following]
--2016-09-30 10:10:20--  https://github.com/area515/Photonic3D/raw/master/host/bin/start.sh
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/area515/Photonic3D/master/host/bin/start.sh [following]
--2016-09-30 10:10:20--  https://raw.githubusercontent.com/area515/Photonic3D/master/host/bin/start.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.16.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.16.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6359 (6.2K) [text/plain]
Saving to: ‘start.sh’

start.sh                                                    100%[========================================================================================================================================>]   6.21K  --.-KB/s    in 0s

2016-09-30 10:10:20 (18.5 MB/s) - ‘start.sh’ saved [6359/6359]

root@odroid64:~# sudo chmod 777 start.sh
root@odroid64:~# sudo ./start.sh
Local Config: /root/3dPrinters/config.properties
Installing RxTx
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  librxtx-java
0 upgraded, 1 newly installed, 0 to remove and 354 not upgraded.
Need to get 154 kB of archives.
After this operation, 927 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports xenial/universe arm64 librxtx-java arm64 2.2pre2-13 [154 kB]
Fetched 154 kB in 0s (860 kB/s)
Selecting previously unselected package librxtx-java.
(Reading database ... 190035 files and directories currently installed.)
Preparing to unpack .../librxtx-java_2.2pre2-13_arm64.deb ...
Unpacking librxtx-java (2.2pre2-13) ...
Setting up librxtx-java (2.2pre2-13) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
W: --force-yes is deprecated, use one of the options starting with --allow instead.
No X server running, starting and configuring one
No protocol specified
xhost:  unable to open display ":0.0"
mv: cannot stat 'cwh-.*.zip': No such file or directory
xauth:  file /root/.Xauthority does not exist

X.Org X Server 1.18.3
Release Date: 2016-04-04
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.13.0-85-generic aarch64 Ubuntu
Current Operating System: Linux odroid64 3.14.65-61 #1 SMP PREEMPT Wed May 25 03:16:39 BRT 2016 aarch64
Kernel command line: root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait ro console=ttyS0,115200n8 console=tty0 no_console_suspend hdmimode=800x480p60hz m_bpp=32 vout=dvi fsck.repair=yes net.ifnames=0 elevator=noop disablehpd=true
Build Date: 18 May 2016  01:12:13AM
xorg-server 2:1.18.3-1ubuntu2.2 (For technical support please see http://www.ubuntu.com/support)
Current version of pixman: 0.33.6
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Fri Sep 30 10:10:47 2016
[... 740 lines cut ...]
  inflating: lib/swagger-core-1.5.8.jar
  inflating: lib/swagger-jaxrs-1.5.8.jar
  inflating: lib/swagger-models-1.5.8.jar
  inflating: lib/validation-api-1.1.0.Final.jar
  inflating: lib/websocket-api-9.3.7.v20160115.jar
  inflating: lib/websocket-client-9.3.7.v20160115.jar
  inflating: lib/websocket-common-9.3.7.v20160115.jar
  inflating: lib/websocket-server-9.3.7.v20160115.jar
  inflating: lib/websocket-servlet-9.3.7.v20160115.jar
  inflating: lib/xalan-2.7.0.jar
  inflating: lib/xml-apis-2.0.2.jar
  inflating: lib/xml-apis-ext-1.3.04.jar
  inflating: lib/xmlgraphics-commons-1.5.jar
Turning off screen saver and power saving
No protocol specified
xset:  unable to open display ":0.0"
No protocol specified
xset:  unable to open display ":0.0"
No protocol specified
xset:  unable to open display ":0.0"
Installing CWH as a service
Determinging if one time install has occurred
grep: /root/3dPrinters/config.properties: No such file or directory
Starting printer host server
`

WesGilster, 2016-10-04:
I'm not sure what you are using to determine Photonic isn't starting. From what I can tell, it's started fine. I don't know much about ODroid. It looks like ODroid doesn't have an xserver started:

> No X server running, starting and configuring one
> No protocol specified
> xhost: unable to open display ":0.0"

Once XWindows is operational, you should be good to go...

kloknibor, 2016-10-04:
You are right, I experimented some more and Photonic3d started working but I couldn't connect to the screen. I will play around some more to get Xwindows working but have some other issues first :( Will close it for now

closed 2016-10-04 by kloknibor