area515/Photonic3D #262: [BUG] USB upload not working

kloknibor, 2016-10-06:
Hi,

A while back wes was saying that automatic USB upload would be possible by adding the following line to config.txt :  

uploaddir=/[usb]/[mount]/[point]

WesGilster, 2016-10-06:
USB upload should work without any changes if this is in your config file:
feature.org.area515.resinprinter.usbimport.USBUploader=true

That line(which is set by default) will make sure that the contents of the USB stick are copied to your uploaddir. If this doesn't work, it's a bug.

Going on to a different subject, if you add this to your config file:
uploaddir=/[usb]/[mount]/[point]    (which must be manually set)

Then your normal uploaddir will be your usb stick. I don't think that is what you are looking for.

kloknibor, 2016-10-06:
Hmm than my thoughts were right...

I have " feature.org.area515.resinprinter.usbimport.USBUploader=true" in my current config file and it didn't work. 

Also adding uploaddir=/[usb]/[mount]/point to make it my permanent upload folder didn't work.

The way I tested this was by using a raspberry pi 3 (Oh shit there we go probably...) and an USB 3.0 stick (This should be backwards compatible with USB 2.0 so won't see any problems there, but to be honest I didn't check to get acces to it over ssh) I will try this tomorrow to test if I can get acces at all, otherwise that would explain a lot. Sorry I should have tested that before but it's been quite hectic lately!

On the USB was nothing else than 2 .stl files.

Robin

WesGilster, 2016-10-12:
Fixed and closed in the latest version. Apparently the listRoots() method doesn't enumerate USBmount points like it does on Windows.

kloknibor, 2016-10-25:
I just tried testing this but it is not working. I uploaded this file 
[Octocat-v2.zip](https://github.com/area515/Photonic3D/files/550621/Octocat-v2.zip) to 2 different USB sticks. (allright one was a micro-SD with USB adapter and the second one was an USB 3.0 datatraveler G4 32GB USB stick).

I used this image to test this : http://d359hvxwrop0le.cloudfront.net/photonic-image.zip (from the README.md)

I didn't saw much usefull in the logs:
[LogBundle (11).zip](https://github.com/area515/Photonic3D/files/550626/LogBundle.11.zip)

Could we reopen this?

WesGilster, 2016-10-25:
Reopened. I'll need to add some debugging to this to determine why it's not working. Then the logs will help.

WesGilster, 2016-10-26:
I confirmed this is working on my pi, but you'll need to test the latest dev branch(WesGilster-6847f42d) to get debugging logs setup for USB.

kloknibor, 2016-10-26:
Since for some reason the latest version isn't set as latest on github how could I do this from start.sh? Or should I just install and throw this zip file on top of it?

WesGilster, 2016-10-26:
James' fix is the latest release here:

https://github.com/area515/Photonic3D/releases

Thanks,
Wes G.

WesGilster, 2016-10-27:
This works fine on my Raspberry Pi, so I've had to add some logging to determine why it might not be working for you. Plug in your USB drive and send me the logs. I'm assuming I need to use a different mount point for your drive, but not really sure.

kloknibor, 2016-11-02:
Sorry for the delay! I just tried this with a version I just build from your repo. Still not working and this is the result in the log : 

2016-11-02 14:43:44,295 DEBUG o.a.r.u.USBUploader$1 [PrintJobProcessorThread-2] MasterRoots:{/=/}
2016-11-02 14:43:44,296 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] listroot:/
2016-11-02 14:43:44,296 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Config root:/media
2016-11-02 14:43:44,296 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Existing Child under config root:/media
2016-11-02 14:43:44,297 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Config root:/storage
2016-11-02 14:43:44,297 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] allRoots:[/]
2016-11-02 14:43:44,297 DEBUG o.a.r.u.USBUploader$1 [PrintJobProcessorThread-2] Unplugged drives:{}
2016-11-02 14:43:47,298 DEBUG o.a.r.u.USBUploader$1 [PrintJobProcessorThread-2] MasterRoots:{/=/}
2016-11-02 14:43:47,299 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] listroot:/
2016-11-02 14:43:47,300 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Config root:/media
2016-11-02 14:43:47,301 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Existing Child under config root:/media
2016-11-02 14:43:47,301 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] Config root:/storage
2016-11-02 14:43:47,302 DEBUG o.a.r.u.USBUploader [PrintJobProcessorThread-2] allRoots:[/]
2016-11-02 14:43:47,303 DEBUG o.a.r.u.USBUploader$1 [PrintJobProcessorThread-2] Unplugged drives:{}

I would like to note that the USB hasn't been unplugged at any point.

kloknibor, 2016-11-02:
complete logbundle : 
[LogBundle.zip](https://github.com/area515/Photonic3D/files/566844/LogBundle.zip)

WesGilster, 2016-11-02:
Ok. So what is the mount point that your USB stick creates under linux? In other words, if you wanted to 'ls' the files from a Unix shell, what would you do?

It doesn't look like there are any drives coming up under "/media".

kloknibor, 2016-11-02:
It seems that it isn't automatically mounting! Should have checked this earler : 

output from lsblk :

NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1 28.9G  0 disk
└─sda1        8:1    1 28.9G  0 part
mmcblk0     179:0    0 14.4G  0 disk
├─mmcblk0p1 179:1    0   63M  0 part /boot
└─mmcblk0p2 179:2    0 14.4G  0 part /

WesGilster, 2016-11-03:
np. Once you get your usb stick working, it should automatically upload associated files.

WesGilster, 2016-11-11:
This looks to be fixed, I've also removed the logging because it was quite excessive. If you want the logging back in, you can add 'usb' to this line in the log4j2.properties:

loggers=hex,serial,printer,media,cleanupCWS,cleanupZip

closed 2016-11-11 by WesGilster

kloknibor, 2016-11-21:
Hi,

I would like the opinion of you guys.

Currently the USB function doesn't work for me since the USB drive isn't automatically mounted. I did some research and found this lib : 

https://usbmount.alioth.debian.org/

with some more backstory found here which I found later : 

http://raspberrypi.stackexchange.com/questions/41959/automount-various-usb-stick-file-systems-on-jessie-lite

I like the function, but the lib is already nearly 10 years old and unmaintained... So do you think it is a good idea to integrate it into our image? Or have better suggestions?

kloknibor, 2016-11-21:
PCManFM also seems to automount but from memory perspective and this being a GUI tool it might not be wanted

WesGilster, 2016-11-22:
I haven't tried this, but udev should be all you need. Here is someone who has tried it:
https://www.axllent.org/docs/view/auto-mounting-usb-storage/

kloknibor, 2016-11-22:
Thanks for the tip! I'm gonna try this straigh away!

https://www.axllent.org/docs/view/auto-mounting-usb-storage/

kloknibor, 2016-11-22:
Works flawlessly! Thank you!

On 22/11/2016 11:30:18, Robin <robinkolk@msn.com> wrote:

Thanks for the tip! I'm gonna try this straigh away!

https://www.axllent.org/docs/view/auto-mounting-usb-storage/