area515/Photonic3D #273: Put printable in the middle of the screen

kloknibor, 2016-10-24:
Hi,

Personally I think it won't make sence to put the printed parts by default in the upper left corner and I think it would be better to center these by default. 

This is because DLP printers have the best resolution in the center and it diminish problems for printers with a deadzone on the edge. This way users don't have to move the part every single time.

Feel free to disagree ofcourse!

kind regards,

Robin

WesGilster, 2016-10-24:
I agree. Are you seeing something different?

kloknibor, 2016-10-24:
On the version I still worked on all parts where printed in the upper left corner. Let me test the new version to be sure ;)!

WesGilster, 2016-10-24:
Yeah that was a few versions ago. Should be fixed now.

kloknibor, 2016-10-25:
Sorry, this is indeed fixed in the latest release!

WesGilster, 2016-10-25:
np.

closed 2016-10-25 by WesGilster