area515/Photonic3D #274: File failed

kloknibor, 2016-10-25:
Hi I tried the new image (http://d359hvxwrop0le.cloudfront.net/photonic-image.zip)

I tried 2 previous working files which are the following files ; 

[printables.zip](https://github.com/area515/Photonic3D/files/550670/printables.zip)

they both failed. here are the logs : 

[LogBundle (12).zip](https://github.com/area515/Photonic3D/files/550671/LogBundle.12.zip)

kind regards

(PS this is running version 1.0.10-beta3)

WesGilster, 2016-10-26:
Those changes were just checked in by James on Oct 22nd and 1.0.10-beta3 was built on Oct 15th. You need to test with a development branch version like WesGilster-6847f42d or something recent.

jmkao, 2016-10-26:
Let me tag a build since I merged the PR already, should've done it before.

jmkao, 2016-10-26:
This is done. The release should auto-update when you next run start.sh.

closed 2016-10-26 by jmkao

kloknibor, 2016-10-26:
Updating to 1.0.11 doesn't resolve the issue :( Stl files still fail straight away and the release from wes is uncompiled, only the source code. 

I will try to compile myself. But we might want to fix it ASAP in our stable repo :)!

kloknibor, 2016-10-26:
Zip files seem to work but .stl not