area515/Photonic3D #275: Build failing

WesGilster, 2016-10-26:
Started getting this on Gradle builds.
23:57:55.186 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] Execution failed for task ':githubRelease'.
23:57:55.189 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] > Cannot invoke method replace() on null object

jmkao, 2016-10-26:
Checking now, in the meantime, if you link Travis CI to your github account and your fork, and then set the GITHUB_ACCESS_TOKEN environment variable in the config to include a personal access token you make at https://github.com/settings/tokens, then you can create releases via tag or from the GitHub UI by clicking the "Draft a New Release".

WesGilster, 2016-10-26:
Cool I'll try that tomorrow thanks

jmkao, 2016-10-26:
Hmm... I can't reproduce this on Linux or Windows. Can you send the output with `./gradlew --stacktrace --info fullRelease`

WesGilster, 2016-10-26:
Is it possible I lost my credentials in my gradle config? Seems strange that would cause this.

[buildFail.txt](https://github.com/area515/Photonic3D/files/552388/buildFail.txt)

jmkao, 2016-10-26:
No, that can't be it. Your credentials and the identification of your repoName out of `..` is required by the `setupRemote` and `setupRelease` tasks, which `githubRelease` depends on and both ran fine. There must be something else wrong...

WesGilster, 2016-10-30:
Got Travis working. I'm not sure if I need to run the build locally now. I'll keep it open for you, if you want to continue to look at it.

jmkao, 2016-10-30:
Yeah, keep it open. I'll poke at it some more and close it if I can't reproduce it.