area515/Photonic3D #277: Print failing - NullPointerException in STLFileProcessor.java

WGAndrew, 2016-10-26:
Getting this when printing a file that's worked in previous builds of cwh. Seems to be failing with a NullPointerException in org.area515.resinprinter.job.STLFileProcessor.processFile(STLFileProcessor.java:70)
This is on a simulated display, so not even using any photocentric-specific display code.
Any ideas please?

`2016-10-26 14:26:05,683 INFO o.a.r.s.PrintableService [qtp4599601-15] Loaded printable file: /root/uploaddir/charmander_starter_1gen_flowalistik.STL
2016-10-26 14:26:07,439 DEBUG o.a.r.p.PrinterManager [qtp4599601-15] Attempting to assign job:charmander_starter_1gen_flowalistik.STL (No Printer) to printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display)
2016-10-26 14:26:07,441 INFO o.a.r.p.PrinterManager [qtp4599601-15] Assigned job:charmander_starter_1gen_flowalistik.STL assigned to printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display) to printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display)
2016-10-26 14:26:07,446 INFO o.a.r.j.PrintJobProcessingThread [PrintJobProcessorThread-3] Starting:charmander_starter_1gen_flowalistik.STL assigned to printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display) on Printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display) executing on Thread:PrintJobProcessorThread-3
2016-10-26 14:26:07,497 INFO o.a.r.s.ZSlicer [PrintJobProcessorThread-3] Load file start
2016-10-26 14:26:07,523 INFO o.a.r.s.ZSlicer [PrintJobProcessorThread-3] Load file stop:26
2016-10-26 14:26:07,524 ERROR o.a.r.j.PrintJobManager$JobCloser [PrintJobProcessorThread-2] Job Failed:PrintJobProcessorThread-2
java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.util.concurrent.FutureTask.report(FutureTask.java:122) ~[?:1.8.0_101]
	at java.util.concurrent.FutureTask.get(FutureTask.java:192) ~[?:1.8.0_101]
	at org.area515.resinprinter.job.PrintJobManager$JobCloser.run(PrintJobManager.java:42) [photonic3d.jar:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0_101]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:1.8.0_101]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:1.8.0_101]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:1.8.0_101]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [?:1.8.0_101]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [?:1.8.0_101]
	at java.lang.Thread.run(Thread.java:745) [?:1.8.0_101]
Caused by: java.lang.NullPointerException
	at org.area515.resinprinter.job.STLFileProcessor.processFile(STLFileProcessor.java:70) ~[photonic3d.jar:?]
	at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:38) ~[photonic3d.jar:?]
	at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:11) ~[photonic3d.jar:?]
	... 6 more
2016-10-26 14:26:07,527 DEBUG o.a.r.p.PrinterManager [PrintJobProcessorThread-2] Attempting to dissassociate job:charmander_starter_1gen_flowalistik.STL assigned to printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display) from printer
2016-10-26 14:26:07,527 INFO o.a.r.p.PrinterManager [PrintJobProcessorThread-2] Disassociated job:charmander_starter_1gen_flowalistik.STL (No Printer) from printer:Photocentric 10(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display:Simulated display)`

[LogBundle (8).zip](https://github.com/area515/Photonic3D/files/553526/LogBundle.8.zip)

WGAndrew, 2016-10-26:
Also happens in latest unmodified raspberry pi image installed from: http://d359hvxwrop0le.cloudfront.net/photonic-image.zip

kloknibor, 2016-10-26:
#274 is the same problem I guess, they should have fixed it yesterday but we might want to pull the latest dev version for now if I understood this correctly

WGAndrew, 2016-10-26:
Sadly, it looks like 10.0.11 doesn't have a fix either:
[image]

WesGilster, 2016-10-26:
Hold on, the fix has nothing to do with STL files. The fix James submitted was designed for a zip of slice files. Are you saying that ALL STL files are no longer operational, or just the above STL file?

James and I are putting together a set of tests for all file processors because we really need to get some stability in this area.

WGAndrew, 2016-10-26:
None of the stl files I've tried have worked in either 10.0.10 or 10.0.11
(I tried 'vanilla' photonic as I wanted to make sure it wasn't something we'd caused with our code additions)

Zips of layer images from creation workshop and slacer do work, but it seems to be at the expense of stl files sadly.

It's 10.30pm over here but I'll try to be able to reply for another half hour to an hour if you need any more info - though I don't have access to the pi until tomorrow now sadly.

WesGilster, 2016-10-26:
If I can reproduce, I'll fix tonight.

WGAndrew, 2016-10-26:
Thanks Wes, I appreciate that :)

kloknibor, 2016-10-27:
Was you able top reproduce the issue? If not is there anyway we can help?

WesGilster, 2016-10-27:
I couldn't check it in last night, but checked in this morning and Travis is running now on Area515. That version should have all of the fixes for printing without Customizers. Seemed like that was causing our pain. Understandably as well since I haven't been testing much in without Customizers. I'll see about getting some tests setup with all the file types so that this is less likely to happen in the future.

WGAndrew, 2016-10-27:
Should also explain that Zipped image sets (from cw or slacer) do work, but at the expense of stls.

(Apologies if there's strange formatting, commenting via email since it's 10.30pm over here)

WGAndrew, 2016-10-27:
All stl files I've tried have failed with that nullpointerexception in both 10.0.10 and 10.0.11...

WesGilster, 2016-10-28:
I'm not sure if we are all on the same page here with testing, so here is a bit of a summary:
1.0.10 - Printing without customizers is broken
1.0.11 - Printing without customizers was fixed for zip of slices (James fixed)
1.0.12 - Printing without customizers fixed for all (Wes fixed)

I can't tell by the comments where your testing is taking place.

kloknibor, 2016-10-28:
Hi wes,

Your "log" is correct and 1.0.12 is working ;)! Andrew was doing an overnight test but the tests I quickly did turned out to be working fine ;)!

Thank you! 

Kind regards,

Robin

WGAndrew, 2016-10-28:
apologies for that Wes - I'd originally replied by email to your GitHub response (a little before posting this comment: https://github.com/area515/Photonic3D/issues/277#issuecomment-256483747) , and apparently there's a huge delay in GitHub processing an email!

WesGilster, 2016-10-28:
NP, I just wanted to make sure it was working. I'll close and take a look at the other bug you opened.

closed 2016-10-28 by WesGilster