area515/Photonic3D #281: Merge area515/octopi changes back upstream

jmkao, 2016-11-06:
@guysoft left a comment on https://github.com/Photocentric3D/OctoPi/commit/7e3d2c8a2d618914ca9e7ce384f7f4d930392fc9 inviting us to submit a PR of our changes back upstream.

This is appealing since it helps keep our variant in sync with any framework changes, although there were some changes to core files that we need to discuss.

Need to decide on how to encapsulate those changes and then merge them upstream.

jmkao, 2016-11-06:
The main core file changes are:
* src/chroot_script - The official repo always installs OctoPrint, presuming that a variant will then modify the OctoPrint distro somehow. Since Photonic3D is completely standalone, it doesn't need OctoPrint at all, so I added a flag (OCTOPI_INCLUDE_OCTOPRINT) that wraps all of the OctoPrint install code and allows it to be supressed entirely
* src/common.sh - Shored up the mount logic, the existing logic can't handle spaces in paths (need quotes around variables) and doesn't like variants that contain "on". These changes are generic and should be able to simply be merged upstream

Questions for @guysoft:
 * Are you okay with the flag that disables the installation of OctoPrint?

WesGilster, 2016-11-06:
@guysoft Hey, thanks for the invitation. That's pretty cool. From your main page it says:
"OctoPi is a Raspberry Pi distribution for 3d printers."

Does that mean you are open to installing printer software outside of OctoPrint? As James mentioned, we are standalone software and it might be cool open your distribution up to the resin printing user base.

guysoft, 2016-11-07:
Hey, so here are answers to the questions:

1. There in a flag to disable OctoPrint installtion now. You can add to your variant config script `OCTOPI_INCLUDE_OCTOPRINT=no`
2. That looks cool! might interest @foosel too. The Offical OctoPi builds I release are born here: http://gnethomelinux.com:8080/job/OctoPi
3. I am open to having variants of OctoPi where OctoPrint is not installed, thats the point of the variant system, you can change it to do whatever you want.

WesGilster, 2016-11-10:
Also James, we may want to scale back the memory usage of Photonic for his image. Once Photonic3d isn't the main software on the Pi, it really should cooperate a bit more nicely with the other large apps.

guysoft, 2016-11-10:
Just to be clear - OctoPi/Photonic3D is a variant. You can decide what will
bw shipped from it. The priduced image would the same that you have now.