area515/Photonic3D #284: Print .cws file failed

vzex, 2016-11-09:
the Status: Failed Your printer configuration isn't capable of this feature
and log files:
[LogBundle.zip](https://github.com/area515/Photonic3D/files/581123/LogBundle.zip)
The printer just stop after lift up and lift down, and i can't find out the reason

vzex, 2016-11-09:
btw,my machine is a lcd dlp printer, and when i test a cws file to print, the lcd screen can be light up(M106 S255), but there is no render pic on screen,so I just wonder whether this host can support this kind of printer?

WesGilster, 2016-11-10:
It looks like you were trying to print a file called: ship.cws. Could you attach the gcode file that is contained in the cws file? That would help me fix this issue. By the look of the code it looks like you have a blank line after the following gcode: M106 S255. 

If you could confirm that, I could fix the bug quite easily and write a test for the situation. As a temporary fix, you might be able to remove the line yourself in Creation Workshop and recreate the cws file. Although I would still like to fix the problem because my gcode parser should not be that picky.

vzex, 2016-11-10:
I'll upload this file later, and there's another problem, the screen doesn't show any pics during printing, and  my printer has a resolution of 2560*1440 lcd connected with pi through the hdmi,Is there any reason cause this?

WesGilster, 2016-11-10:
When you say "the screen" doesn't show any pics while printing. Do you mean that it doesn't show up on the lcd screen or it doesn't show the printed picture on Photonic3d itself, or both?

WesGilster, 2016-11-10:
If it shows up on Photonic3d but not on the LCD printer screen you probably have a screen config issue on the pi.

vzex, 2016-11-10:
I mean, doesn't show the printed pic,and I upload a .cws here,It's not the original file I used, because it't at home now, so I made a new slice using the same stl file(the slice configure is more or less the same), you could check whether this file can [use.](https://drive.google.com/open?id=0B_f34qEeuI80UVo0RTZMR3l4WjA)

WesGilster, 2016-11-10:
"Doesn't show the printed pic" is an ambiguous statement and doesn't help me to determine whether you can't see the picture on the Photonic3d html page, the printer's LCD screen or both of them.  Depending on how you answer that question, the solution could be this:
https://forum.photonic3d.com/t/upgraded-to-1-0-5-no-longer-able-to-print-cws-files/73/10?u=wesgilster

I've replicated your other problem, built a test and a dev release should be available in the next few minutes:
[image]

vzex, 2016-11-10:
Ok, i'll talk aboue the screen problem later, so , is the .cws problem sloved

WesGilster, 2016-11-10:
Yes, fixed yesterday in the dev Repo, but it is not in production Repo yet.

Thanks,
Wes G.

vzex, 2016-11-10:
So I could use the "Latest development" to re-install it ?

WesGilster, 2016-11-10:
Yes, but keep in mind that every time you reboot your pi, it will 
attempt to reinstall from production which will wipe out these changes 
until the changes are promoted to production. I'll see if I can get that 
done today.

vzex, 2016-11-10:
OK, you're so sweet!

vzex, 2016-11-10:
pi@LCHR:~ $ sudo ./start.sh WesGilster
Local Config: /root/3dPrinters/config.properties
mv: cannot stat ‘cwh-.*.zip’: No such file or directory
Checking for new version from Github Repo: WesGilster/Creation-Workshop-Host
Local Tag: 1.0.13
Network Tag: WesGilster-783dbf7
Installing latest version of cwh-: WesGilster-783dbf7
rm: cannot remove ‘/tmp/’: Is a directory
http://: Invalid host name.
wget of  failed. Aborting update.

update failed

vzex, 2016-11-10:
[image]
hi, here is the second question, I setup the machine just like this, without choosing "Projector Serial Port",because I just want the pic render to pi's hdmi port(I think this looks like in creation workshop).After setup the printer started, and just show nothing on screen during print.So did I choose the right option?(leave the Projector Serial Port blank)

kloknibor, 2016-11-10:
Yes you don't need to set the serial port for the projector. This is only useful if you have connected your projectors serial port to the USB on the raspberry pi. When you to this you would be able to do things is powering your projector on and off, getting bulb hours etc.

vzex, 2016-11-10:
[image]
[image]
Now my printer behavior like this, during my test, the lcd just looks like the projector control(Show Blank) in windows's creation workshop. If i connect the printer to the pc use creation workshop, the printer works fine.And I can't control the projector in the webpage either.
[image]
I just want to make sure it's the photonic3d's problem, or my pi' config problem?

WesGilster, 2016-11-10:
You shouldn't be able to control the projector or detect a projector because you don't have one. Projector controls are things that Kloknibor was talking about querying bulb hours and auto power controls. Since you have an LCD screen things just work "automatically" There really isn't much Photonic needs to do.  Have you tried to show the grid screen? If the grid shows up, then you know Photonic can show something on your screen.

vzex, 2016-11-11:
Yeah you're right,it's my problem

closed 2016-11-11 by vzex

WesGilster, 2016-11-11:
I wasn't implying it was "your" problem. I was trying to determine the source of the problem. Are you sure you don't need further help on this issue?

vzex, 2016-11-11:
yes, after I did some tests, I found the screen works well with a small resolution, then I realized it's a setting problem of /boot/config.txt.

vzex, 2016-11-11:
If I have other questions, I'll open another issue, thanks for your patience