area515/Photonic3D #286: raspberry pi really support 2k hdmi output?

vzex, 2016-11-11:
Actually, after I made a lot of tests, I just couldn't let the screen show images.So I just wonder if  pi really support high resolution？What I got is blank screen.
I have some config of /boot/config.txt below
`disable_overscan=1
framebuffer_depth=24
gpu_mem=192
framebuffer_ignore_alpha=1
start_x=1

hdmi_ignore_edid=0xa5000080
hdmi_cvt 2560 1440 50 3 0 0 1
hdmi_group=2
hdmi_mode=87
hdmi_pixel_freq_limit=300000000
hvs_priority=0x32ff
max_framebuffer_width=2560
max_framebuffer_height=1440
framebuffer_width=2560
framebuffer_height=1440
config_hdmi_boost=4
hdmi_force_hotplug=1
`

and tvservice -d shows:

- HDMI:EDID version 1.3, 1 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is V_R__________ 
HDMI:EDID found monitor S/N descriptor tag 0xff
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 23-75 Hz, horizontal is 15-240 kHz, max pixel clock is 300 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID established timing I/II bytes are 00 00 00
HDMI:EDID standard timings block x 8: 0x0000 0000 0000 0000 0000 0000 0000 0000 
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:4
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID failed to find a matching detail format for 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID calculated refresh rate is 50 Hz
HDMI:EDID guessing the format to be 1440x2560p @50 Hz
HDMI:EDID found unknown detail timing format: 1440x2560p hfp:60 hs:10 hbp:35 vfp:4 vs:2 vbp:2 pixel clock:198 MHz
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID does not understand CEA format code 0
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48|88|96|176|192 kHz, sample size: 16|20|24 bits
HDMI:EDID found HDMI VSDB length 6
HDMI:EDID HDMI VSDB has physical address 3.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID adding mandatory support for DMT (4) 640x480p @ 60Hz
HDMI:EDID adding mandatory support for CEA (1) 640x480p @ 60Hz
HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
HDMI:EDID no known preferred format has been set
HDMI:EDID filtering preferred group has been changed from Invalid to CEA
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 61864)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 18432
HDMI:EDID preferred mode is updated to CEA (1) 640x480p @ 60 Hz with pixel clock 25200000 Hz
HDMI:EDID has HDMI support and audio support

WesGilster, 2016-11-11:
That seemed to be quite a high resolution for the Pi, but I can't really tell you. We've got a few Pi experts here, but you may actually want to ask your question on the Raspberry Pi forums as well. Here is a topic from the Pi forums that my help you out:

https://www.raspberrypi.org/forums/viewtopic.php?f=7&t=307

jmkao, 2016-11-11:
It should be possible. The RPi can handle up to 4K at 15hz refresh. Choppy for desktop use, but plenty fast for printing.

Can you paste the output of `tvservice -s`?

vzex, 2016-11-11:
tvservice -s
state 0x120006 [DVI DMT (87) RGB full 16:9], 2560x1440 @ 50.00Hz, progressive

WesGilster, 2016-11-11:
It looks like you need to dial back your refresh rate to 15Hz and see what happens.

vzex, 2016-11-12:
ok,I'll make a test after printing.

vzex, 2016-11-13:
I'm success with 2560*1440@50hz output,but when I make the Grid Screen test,the image on screen is not fullscreen.
[image]
[image]

So  is there any other params need to be set ?
When I's testing, I found the resolution in photo3d
[image] is this right?

my tvservice -s now  : 
state 0x120006 [DVI DMT (87) RGB full 16:9], 2560x1440 @ 50.00Hz, progressive

vzex, 2016-11-13:
I think it's maybe a rotate problem, I'll check it

WesGilster, 2016-11-13:
That doesn't look right. When you calibrate the screen those values will be detected from your hardware. Otherwise you can override those settings in the printers page here:
[image]

kloknibor, 2016-11-16:
it is totally possible. We have got 1K (and lower) 2.5K and 4K screens to work around here! We had to set the things in boot/config.txt right our settings for the 2.5K screen were : 

overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
#disable_overscan=1
hdmi_drive=1
hdmi_ignore_edid=0xa5000080
hdmi_group=2
hdmi_mode=87
#60Hz
hdmi_pixel_freq_limit=206000000
hdmi_timings=2048 0 150 5 5 1536 0 3 1 9 0 0 0 60 0 205210000 1
max_framebuffer_width=2048
max_framebuffer_height=1536
display_rotate=0
framebuffer_width=2048
framebuffer_height=1536

I would suggest to bring you refresh rate down as already was mentioned ;)!

vzex, 2016-11-17:
@kloknibor Actually my config looks the same as your's, but i don't know  how to set the hdmi_timings, even I 've got the datasheet - -!. I can't figure out which argument cause the screen just show 1/2

vzex, 2016-11-23:
problem solved

closed 2016-11-23 by vzex