area515/Photonic3D #287: printer not starting : internal server error

kloknibor, 2016-11-16:
Hi,

I just had the following error on an clean install, it might be due our costum version so I asked WG as well. But it looks like a photonic error. What happens is that I just installed the profile from github (which we used before) but when I started the printer I got the error "internal server error" and wouldn't start when I took a look at the printer configuration I couldn't pick a single display device.

[LogBundle (6).zip](https://github.com/area515/Photonic3D/files/595494/LogBundle.6.zip)

WesGilster, 2016-11-17:
This is definitely due to Javascript going a bit wild in your custom 
version. The logs are showing some serious spamming of the 
"http://localhost:9091/services/printJobs/getByPrinterName/" method. 
Perhaps the Javascript has gotten caught in some refresh loop of some sort?

Thanks,
Wes G.

kloknibor, 2016-11-17:
Thank you for taking a look anyway! We resolved the issues ;)!

closed 2016-11-17 by kloknibor