area515/Photonic3D #290: Feature request : Change hostname easily for bonjour

kloknibor, 2016-11-18:
Hi,

I would love an easy function to change hostname, currently I user a shell script I made to do this which can be found here : 

`
#Get new hostname
NEWHOSTNAME="$INPUT_FROM_WEB"

# get current hostname
HOSTNAME=$(hostname)

# set new hostname
sudo sed -i -e 's,'"$HOSTNAME"','"$NEWHOSTNAME"',g' /etc/hosts
sudo sed -i -e 's,'"$HOSTNAME"','"$NEWHOSTNAME"',g' /etc/hostname
sudo /etc/init.d/hostname.sh
sudo reboot

`

this should be safe as long as people not first change the name to localhost and than chance it to something else. This will break /etc/hosts and our custom version. But I don't see that happen soon.

Would it be possible to get the var $INPUT_FROM_WEB from the settings page and call this script? Or maybe it's nicer to call it from java. I will leave that in the middle :)

Let me know what you guys think!

WesGilster, 2016-11-28:
I've been thinking about putting together a mechanism to talk to the OS through Photonic. I don't have anything concrete setup yet though.

kloknibor, 2016-11-29:
Alright sounds cool! 

I'm not sure how this can be done in a safe way since you don't want to run a bash script from a webpage with variable input because of security risks I suppose. But honestly I'm not sure how else this can be achieved. In the end we do need to alter files in the OS. I'm to unexperienced to know how to do that safe to be honest :p Looking forward to see your implementation and start reading it :)!

WesGilster, 2016-11-30:
The fact that we allow Java API access through both gcode templates and Javascripting means that we already give everyone access to the underlying OS and more, it's just not as obvious to exploit as this would be. However, that's not really a problem because access to this feature will be governed by this annotation on the Java method:
`@RolesAllowed({PhotonicUser.FULL_RIGHTS, PhotonicUser.OS_EXECUTION})`

That means you will need to have full rights or OS execution permissions to execute that method. This annotation is only respected if users are being authenticated though:

```
config.properties
useAuthentication=true
```

closed 2017-02-20 by kloknibor