area515/Photonic3D #291: Paused with error : Can't seem to find why the error was

kloknibor, 2016-11-18:
Hi,

Last night I had an error on one of the printers. Software was still up and running but the printjob gave an error like something like : Paused with warning.

I couldn't find in cwh.log why it decided to pause the print... Any idea why the print failed?

[LogBundle (4).zip](https://github.com/area515/Photonic3D/files/599841/LogBundle.4.zip)

WesGilster, 2016-11-18:
You really need to use the latest version of Photonic 3D as these logs are getting completely obscured by the USB debugging we put in a while back. In fact, the logs rolled by the time you did the diagnostic dump and it didn't expose the cause of the error.

There are three ways that Photonic3D can be paused due to an error:
1. Printer sent back an error code.
2. InkDetector threw an exception. I'm guessing you aren't even using this.
3. Errors in Geometry while slicing. If you weren't printing an STL, this didn't happen.

If it's not 2 and 3, then this says your printer sent back an error code and there wasn't a description attached to it.

WesGilster, 2016-11-18:
Actually I did another quick check. It looks like you were trying to print gyrothecube.STL  That puts case 3 back into play.

WesGilster, 2016-11-18:
My latest update adds error descriptions for cases 2 and 3 now when PausedWithWarning is set. Case 1 was already being set.

kloknibor, 2016-11-19:
Hi wes,

I agreed with WG that we first needed to fix our issues before merging. But those have been resolved friday so I will do a merge on monday ;) 

Indeed I was printing that .stl (gyrocube.stl) I don't have acces to the file atm but I will check on monday. Shall I send you the file as well?

Thanks!

kloknibor, 2016-11-19:
And indeed I don't use the ink detector

WesGilster, 2016-11-19:
WG is right, mixing too much code from different places can be a 
nightmare if you haven't solved your own issues first.

Thanks,

Wes G.

closed 2016-11-23 by kloknibor