area515/Photonic3D #292: Console error + can't change resin profile

kloknibor, 2016-11-21:
Hi Wes,

We have an error in the developer consoler of chrome. This is what it states : 

```
ReferenceError: printer is not defined
    at refreshSelectedPrinter (printers.js:16)
    at printers.js:49
    at angular.js:10072
    at angular.js:14567
    at n.$eval (angular.js:15846)
    at n.$digest (angular.js:15657)
    at n.$apply (angular.js:15951)
    at l (angular.js:10364)
    at F (angular.js:10536)
    at XMLHttpRequest.K.onload (angular.js:10477)
```

Also when changing slicing profile it looks like you selected the new one from the drop down menu. But after a refresh it will return to the old slicing profile and hasn't changed to the new resin profile.

Both tested in the latest release of area515 :)!

WesGilster, 2016-11-21:
Problem 1 fixed and checked into the dev branch.
The second issue is normal functionality. You need to save your printer 
before those settings take effect.

Thanks,
Wes G.

kloknibor, 2016-11-21:
ah you are right. Since there wasn't a save button around there I assumed it was allright. Sorry!

closed 2016-11-21 by kloknibor