area515/Photonic3D #293: 2k screen,show grids not correct

vzex, 2016-11-22:
Hi, it's me again, after almost two weeks, finally i got my 2k lcd screen show correctly(1440*2560).But here's another problem, when i test the grid show(in the photonic3d's control  page), the grids on screen are not squares but rectangles.
[image]
No matter I set the x width/y width to 1440/2560 or 2560/1440, the result is the same.
[image]
So could you tell me where is the problem?
tvservice -s also shows : 1440*2560 @ 50hz
fbset -s is 1440*2560 two.

vzex, 2016-11-22:
maybe my screen need a rotate 90 degrees in config.txt?

WesGilster, 2016-11-23:
Good work getting your screen working. The grid that you are calibrating will always using the same number of pixels vertical as it does horizontal. This number is denoted by the 100pixels in your screen shot above, and could look like a square or a rectangle depending on the PPI of your LCD screen. If you end up seeing a rectangle, that means your LCD screen has a different X Density than your Y Density. If you know the precise vertical and horizontal PPI from your manufacturer, you can enter those settings manually from the printer screen. If you don't know them, you can calibrate your printer with the calibration process pictured in your screen shot. This will never change the "rectangle" on your screen because your screen can never change it's PPI. However, it will simply allow certain PrintProcessors to understand how to warp the image based upon your LCD screen.

vzex, 2016-11-23:
Hi,I got the machine worked, thanks for your support.The solution is : change 50hz into 48hz,and everything works well!

closed 2016-11-23 by vzex