area515/Photonic3D #294: how to add slicing and ink profiles

kloknibor, 2016-11-24:
Hi,

Is there any easy way to add (and than also remove) slicing and ink profiles? an API for such would be great :)

Kind regards

WesGilster, 2016-11-24:
Done. I didn't realize I didn't have those... I added backend CRUD RESTful methods for slicing profiles and machine configs into the MachineService. Also added a test for these new methods.

On the other hand, many InkConfigs can already be contained and are specific to a single SlicingProfile and can be manipulated completely on the GUI side. That is the same way CW organized it's file structure, and that setup actually seems to make sense. Let me know if there are any RESTful changes required in that area as I really don't see it.

kloknibor, 2016-11-25:
Thanks Wes! 

You are awesome :)! I agree with you only the slicingprofiles will be fine and swagger looked nice ;)!

thanks a lot!

kloknibor, 2016-11-25:
you might have created some errors with your latest release ;)! :  

[image]

closed 2016-11-25 by kloknibor

WesGilster, 2016-11-25:
Ouch didn't see that. New build running.