area515/Photonic3D #295: Ink Configs, and how to select them

WGAndrew, 2016-11-25:
Not necessarily a bug, but wanting to pick your brains.

What I have been doing in the past is using 1 single ink config in a printer profile, then programatically changing cure times and z-heights to make a 100micron slice cure for longer than a 25 micron slice - for example.

Obviously that's fine for one printer, but not great for multiples, and rather than having all this set by Javascript, for ease of updating (say, if there's a new resin formula, or new colour added), it feels like this really should be data driven.

I know that InkConfig supports an array profiles (and that Photonic accepts the array) - so for example, here's a populated set on the Photocentric-Dev branch:
https://github.com/Photocentric3D/Photonic3D-Dev/blob/master/host/printers/photocentric%2010.json

But what I can't seem to do is make a selection 'stick'
Using /apidoc, JSON calls from printflow, or even Photonic3d's web interface doesn't seem to ensure that a chosen InkConfig is then actually read and used by Photonic. I assumed it'd be driven either by "selectedInkConfigIndex": 0 or "SelectedInk": "Firm Amber 100 Micron" - but whenever I try and change these values (either the index, the selectedInk, or both to a matched pair) - Photonic3D seems to just default to 0, and the head InkConfig.

I haven't had a lot of chance to dig with fixing bugs in our changes and implementation, so I figured i'd be better off picking your brains where I should be looking in the java code to investigate and/or fix for that one.

Thanks guys.

WesGilster, 2016-11-25:
You should be able to set the ink configuration by the "SelectedInk". The selectedInkConfigIndex attribute is just a convenience attribute to find the index, this allows the GUI to select the proper config without having to loop through the configurations and find the correct one by name. That really helps when dealing with Angular.

You'll notice that when a SlicingProfile gets serialized back into Java from json, the selectedInkConfigIndex has no way to propagate it's value back into java since it's an empty method in org.area515.resinprinter.printer.SlicingProfile:

```
	private void setSelectedInkConfigIndex(Integer selectedIndex) {
	}
```

Also notice that getSelectedInkConfigIndex() is transient from the standpoint of XML(doesn't get saved to disk) but it's still a good json property. 

If you know this isn't working, I'd be glad to modify this test: org.area515.resinprinter.services.MachineServiceTest

to make sure that json can make a round trip to XML and back again. (I should probably have that test anyway...)

WesGilster, 2016-11-26:
Robin sent another email on this, so I decided to write a test for it and checked into dev:
`org.area515.resinprinter.services.MachineServiceTest.roundTripJSONToObjectToXMLToObject()`

This makes sure that the object keeps its selected ink setup as it gets marshaled back and forth from json and XML to java objects.

I used your file as the input for the test, but it's trying to set the "SelectedInk" to "default" for some reason. That's not the name of an ink configuration and will not work as I described in my prior response. I changed it to "Firm Amber 50 Microns" and it seems to work fine. Let me know if I can help further.

WGAndrew, 2016-11-28:
hmmm - that's odd. Photocentric 10 shouldn't be using default at all [the change is here for that one: https://github.com/Photocentric3D/Photonic3D-Dev/commit/edbf10393602a6e601bf3c53c539595f1a7c7410 ] - so I gues it depends when you grabbed it.

LC HR does still, for some strange reason. Time to kill that ;)

closed 2016-11-28 by WGAndrew