area515/Photonic3D #296: Something's broken in resourcesnew/bootstrap directory

Mattasmack, 2016-11-27:
I just restarted my printer after being away for the weekend, and after Photonic3D updated itself to 1.0.014 the web interface wouldn't load.  I saw no errors in the logs in /opt/cwh on the Raspberry Pi, but the web console on my browser had a bunch of errors related to angular.  I discovered that the directory resourcesnew/angular/js was missing some of the referenced files, and resourcesnew/bootstrap didn't have anything in it at all except for a css directory and something broken named ',.'.

I decided to nuke everything and reinstall from scratch before posting here, and that seems to have fixed most of the problem -- the missing angular files are there now, and the web interface is working and I can print again. (whew!)  Looks like I just got a bad download the first time.

But, the bootstrap directory is still boned (no js subdirectory; something named ',.'), and the web console still shows errors from trying to load bootstrap/js/bootstrap.min.js .  It doesn't seem to be a critical error, at least for the functions of Photonic3D that I use, but something in there is broken.

WesGilster, 2016-11-27:
I suppose it wouldn't be horribly difficult to start performing sha hashes to ensure good downloads, but I'm not confident that is the problem. A corrupted download shouldn't even extract. 

I'm also more than a bit surprised that you are able to function without being able to load bootstrap.js. That's a pretty critical gui and photonic must look absolutely awful without it. I'm not quite sure how anything can work. 

Is it possible that you didn't run the start.sh with sudo? Have you tried a shift-refresh in the browser? When you list the contents of the /opt/cwh/resourcesnew/bootstrap/js directory on disk, does it still not show the contents below:
[image]

Notice that I just performed a new install as well and I'm not receiving the same errors.

Mattasmack, 2016-11-27:
I am sure I used sudo when running start.sh ... I just went back through my command history to check.
I nuked the previous installation by moving start.sh into /opt, deleting the entire /opt/cwh directory, then running sudo start.sh.  In fact, when I did that and found that the second installation seemed to have all the needed angular files but bootstrap was still messed up, I deleted the directory again and did a third installation.  At least from a cursory glance, the second and third attempts gave the same results.  This is what my bootstrap directory has: 

[image]

You know, I guess I don't know what Photonic3D is supposed to look like!  This is what I have now:

[image]

Note the failure to load bootstrap.min.js in the web console at the bottom.  I've never opened up the console on this page before, so I have no idea what errors were or weren't there in previous versions, but the look of the page has been pretty consistent for me, I think.  Is it supposed to look different?

WesGilster, 2016-11-28:
Wow, that's odd. I'm assuming the script went crazy and created some strange directory. The camera.mp4 is normal if you aren't streaming. The look of Photonic also looks normal. 

The bootstrap.min.js is not normal and neither is your ".," directory. That's odd... 

I kindof need to see what happens on the install. There should be some startup logs that show how the installation proceeded when photonic was restarted. I'll need to see those installation they are only in the diag dump immediately following the install.

WesGilster, 2016-11-28:
Also, make sure that directory is deleted before you perform the next Photonic install.

Mattasmack, 2016-11-28:
Got it, tomorrow I'll reinstall and grab all the logs I can find.  (I'm running a print at the moment.)

Mattasmack, 2016-11-28:
Well, I don't know.  I deleted the directory and reinstalled Photonic again, and this time everything looks fine!
I did find that I had to reboot and run fsck to completely clean out the old directory, though.  My current theory is that yesterday when I was deleting the cwh directory between installs, I never bothered to check that everything was, in fact, deleted.  So the broken cwh/resourcesnew/bootstrap directory stuck around between installs and I just never noticed.
Still no idea what caused it in the first place, but it more likely was some transient problem with the computer, not Photonic.  Sorry for the false alarm!  This can probably be marked closed; I'd do it except I have no idea how github works.
And thanks for all your help!

WesGilster, 2016-11-29:
No problem at all, it would be interesting to know how that directory was created in the first place though...

closed 2016-11-29 by WesGilster