area515/Photonic3D #297: CWS files upload and then fail to populate Printables

WGAndrew, 2016-11-28:
Attempting to upload a CWS file results in nothing being added to the Printables tab in Photonic. 
`2016-11-28 15:15:29,724 ERROR o.a.r.j.CreationWorkshopSceneFileProcessor [qtp32195482-14] Unable to open uploaded zip file
java.util.zip.ZipException: zip file is empty
	at java.util.zip.ZipFile.open(Native Method) ~[?:1.8.0_65]
	at java.util.zip.ZipFile.<init>(ZipFile.java:219) ~[?:1.8.0_65]
	at java.util.zip.ZipFile.<init>(ZipFile.java:273) ~[?:1.8.0_65]
	at org.area515.resinprinter.job.CreationWorkshopSceneFileProcessor.zipHasGCode(CreationWorkshopSceneFileProcessor.java:314) [photonic3d.jar:?]
	at org.area515.resinprinter.job.CreationWorkshopSceneFileProcessor.acceptsFile(CreationWorkshopSceneFileProcessor.java:49) [photonic3d.jar:?]
`

Present in v1.0.14

Testing files and log file attached

WGAndrew, 2016-11-28:
[photonic bug.zip](https://github.com/area515/Photonic3D/files/616732/photonic.bug.zip)

WesGilster, 2016-11-29:
This is a bad file(no attached gcode). No print file processor is willing to take responsibility for attempting to print it. Here is the response the browser is getting from the PrintableService when you attempt to upload. I fixed the content type of the return type since it isn't being sent properly, but the response is being ignored anyway.

[image]

WesGilster, 2016-11-29:
Latest version shows error messages coming back from uploads

WesGilster, 2016-11-29:
Here's an example:
[image]

closed 2016-11-29 by WesGilster