area515/Photonic3D #300: [bug] not always being properly connected to the motor controller

kloknibor, 2017-01-06:
Hi,

I've got an issue where the software thinks it is properly connected and I can control the screen. But when I try to send commands manually it doesn't reply on the command and just won't do anything. After stopping and restarting the printer everything is fine again. This is incidentally and not all the time.

[LogBundle (16).zip](https://github.com/area515/Photonic3D/files/690372/LogBundle.16.zip)

WesGilster, 2017-01-07:
There was already some work in this area. Your fork is not up to date with the latest version of development. You'll need to pull and rebuild from development first.

WesGilster, 2017-02-13:
Closing for now since this is related to a bug that is fixed in development.

closed 2017-02-13 by WesGilster