area515/Photonic3D #303: [BUG] first layer mess up time estimation

kloknibor, 2017-02-20:
Hi,

I think it is an pity the first layers of an print mess up the complete estimated time for the rest of the print. Would it be possible to, in case you are using a static movement/.cws, to don't use the first layers for time estimation? (preferably even calculate the complete expected time for a .cws which should be calculatable by the gcode file)

Kind regards,

Robin

WesGilster, 2017-02-21:
You could certainly do that. The problem is that everyone has their own ideas of how to predict the end of the world, but so far I haven't seen anyone do a very good job of it. My theory thus far has been pretty simple. The server side is responsible for supplying the frontend a set of facts about a print job's progress. These facts currently includes things like:
   totalSlices
   currentSlice
   currentSliceTime
   averageSliceTime
   startTime
   elapsedTime
   totalCost
   currentSliceCost

You are attempting to take a very specific case(cws files) and build a rather complicated theory that may or may not end up being that accurate.

WesGilster, 2017-06-28:
Did you ever come up with a way you wanted to do this?

kalvisg, 2019-07-09:
Calculated time will always be easier to calculate but each Raspberry Pi or any other computer will finish at different times which cannot be calculated any more