area515/Photonic3D #304: [BUG] - empty notification windows Wes version

kloknibor, 2017-02-20:
[image]

Hi , in the development version of wes I get the following blank notification when uploading an .cws file. Got this with all .cws files i tried, I attached one : 
[Ag test robin.cws.zip](https://github.com/area515/Photonic3D/files/788132/Ag.test.robin.cws.zip)

WesGilster, 2017-02-21:
I'm not able to reproduce this. I have noticed that when I started floating between different versions of photonic and different repos that there was some fairly strange functionality that would occur if I didn't dump my browser cache. If this doesn't work for you, I'll need to see a log dump and perhaps see the last network responses in the browser debug mode. I'll check a couple more things to see if I can recreate this.

kloknibor, 2017-03-02:
I think this is something in my own front-end. It is an error notification which is not filled. I see I used my own front-end on top of your front-end in the screenshot) (Which I thought I didn't)

With the latest merge of your repo I have the same issue now, So if you can't reproduce it is an problem for me ! Sorry to bother...

closed 2017-03-02 by kloknibor