area515/Photonic3D #306: [BUG] loading bar not showing smoothly but all over the place

kloknibor, 2017-03-02:
Hi,

How is it possible that the loading bar when uploading a part is all over the place? (as in only left and right :P but it certainly is not at the right location!) 

I might want to take a look into this myself but if it is a really small thing or any advice than you would be more than welcome!

Kind regards,

Robin

WesGilster, 2017-03-03:
There is a strong probability that your custom front end has some css issues. Get into debug mode and move the loading bar around until it fits nicely where you want it. Then make a style out of your debugging steps. That's a complete hack, but it might give you some hints as to how things got a bit messed up.

HTML and CSS are extremely easy to use, but surprisingly difficult to master.

kloknibor, 2017-03-03:
Sorry I think I described my issue wrong. I took a closer look just yet... It is this class but it doesn't make sense : 

<div class="progress-bar progress-bar-success progress-bar-striped" role="progressbar" id="progressbar" aria-valuenow="90" aria-valuemin="0" aria-valuemax="100" style="width: 89%;">89%</div>

(the bar was at 89%)

But the bar basically gets too wide or to small after coming to the right position... Samething with the normal frontend... (This was tested with a complete copy of your java script code for confirmation) but it seems less worse... Just not smoothly, I will take a look myself first :)!

closed 2017-03-03 by kloknibor