area515/Photonic3D #307: [New feature] Preview allow the costumizer or at least the previewer on .cws

kloknibor, 2017-03-02:
Hi,

With .zip files, we support the first layer preview and some customizations. I understand that some customizations are not possible (as starting from layer x) but even just a preview would be pretty nice so you know what you are gonna print!

Since a .cws is basically a .zip file with gcode added (and some other useless files) this should be a small thing right?

Kind regards,

Robin

WesGilster, 2017-03-03:
Yeah, seems pretty easy. I'm surprised that it doesn't already work by default.

kloknibor, 2017-03-03:
I'm trying to find out where the customizer decides it doesn't want to show the image for files with the friendly name creationworkshop (so basically an .zip with .gcode in there is that correct? so the extension doesn't really matter?)

I do think this is the right piece of code but I'm not fully understanding it (this code is from the zipfile processor): 

@Override
	public BufferedImage renderPreviewImage(DataAid dataAid) throws SliceHandlingException {
		try {
			prepareEnvironment(dataAid.printJob.getJobFile(), dataAid.printJob);
			
			SortedMap<String, File> imageFiles = findImages(dataAid.printJob.getJobFile());
			
			dataAid.printJob.setTotalSlices(imageFiles.size());
			Iterator<File> imgIter = imageFiles.values().iterator();
	
			// Preload first image then loop
			int sliceIndex = dataAid.customizer.getNextSlice();
			while (imgIter.hasNext() && sliceIndex > 0) {
				sliceIndex--;
				imgIter.next();
			}
			
			if (!imgIter.hasNext()) {
				throw new IOException("No Image Found for index:" + dataAid.customizer.getNextSlice());
			}
			File imageFile = imgIter.next();
			
			SimpleImageRenderer renderer = new SimpleImageRenderer(dataAid, this, imageFile);
			RenderedData stdImage = renderer.call();
			return stdImage.getPrintableImage();
		} catch (IOException | JobManagerException e) {
			throw new SliceHandlingException(e);
		}
	}
It seems that the previewer first unpacks the whole .zip file and loops through all slices, I was wondering why but this probably has to do with customizers correct?

Many people use this preview also to just properly see what they will be printing. It might be cool to just show the first slice as soon as it is available and use a loading bar for the costumizer? Or even cache the first slice of all files and show that directly while in the background looping through?

WesGilster, 2017-03-04:
Normally, I'd just tell someone right where to go, but I feel like your wanting to learn as well. A good developer sometimes has to be a bit of a detective. So, in this case think about what information you know for sure, and work backwards into the code. 

In this case your only hint towards the crime are "files don't support image preview." See what happens when you search for that phrase in the repo. That search will basically give you everything you need to know.

Spoiler:
CreationWorkshopSceneFileProcessor doesn't implement Previewable.

kloknibor, 2017-03-06:
Haha thanks for that :)! Yeah I was mainly looking for an hint ;)!

I noticed the difference between the ZipImagesFileProcessor.java and saw it missed the class described above which seems the correct code.

So I just added it and I'm gonna test... It had 2 complaints : 

1. FindImages class was not defined... So I added it but it didn't go away. Untill after an hour I found out I need to save the file to refresh the error log... Stupid me :P 

2. Description	Resource	Path	Location	Type
The method renderPreviewImage(AbstractPrintFileProcessor.DataAid) of type CreationWorkshopSceneFileProcessor must override or implement a supertype method	CreationWorkshopSceneFileProcessor.java	/photonic3d/src/main/java/org/area515/resinprinter/job	line 288	Java Problem

This seems since it has nothing to override, I'll look into this now :)!

kloknibor, 2017-03-06:
Hmm now I take a closer look at your hint I need to implement previewable as well I suppose... Since ZipImagesFileProcessor extends CreationWorkshopSceneFileProcessor I could just get this too the CreationWorkshopSceneFileProcessor probably... Lets try :D

kloknibor, 2017-03-06:
ok I take that back now I took a proper look... Sorry

kloknibor, 2017-03-06:
yeah! It works :)!

kloknibor, 2017-03-06:
I did some cleaning up, would you mind taking a look if you agree?

kloknibor, 2017-03-06:
Did one big fix already so the changes I did are : 
https://github.com/Photocentric3D/Photonic3D-Dev/commit/fa5eabddf0f0542f20f11acf8ec573e8f14cb4d0

https://github.com/Photocentric3D/Photonic3D-Dev/commit/0ee950c7e128075fb75fe2e0932453a9f57cac53

WesGilster, 2017-03-07:
Awesome. Nice job! Send a pull request and I'll merge it in.

WesGilster, 2017-03-07:
Pull request to my dev repo that is...

WesGilster, 2017-06-28:
Pulled from Robin's repo a while back.

closed 2017-06-28 by WesGilster