area515/Photonic3D #308: (Question)How to set up my DLP printer in photonic3D?

Jolly2015, 2017-03-04:
Sorry I am a new user of DLP machine. I installed Raspberry Pi System Image. now I can visit dashboardPage, it's shows need to setup a printer,but I don't know the method, is there any instructions for this?

WesGilster, 2017-03-05:
That's it. You've done it. Now it's time to setup a printer or use a community profile. You don't really even need the windows install, since it seems like you already know how to get to the dashboard without it. The windows install is just for those that don't know how to browse to the dashboard.

Not sure where you want to go next, but here are some docs if you are interested:
https://wiki.photonic3d.com/doku.php

Jolly2015, 2017-03-05:
@WesGilster,thanks for your fast answer,I would like to know how to set up a printer next,beacuse there is no  printer shows on the dashboard page even I connected my machine.I try to flowing the instruction(from your link at "plug and play" part) , just a red Photonic3D picture come out,then open a dashboard page still without any prints,should I do something change on my raspberry pi? How Photonic3D communicate with my ramps1.4?How it know the bandrate and serial port number?

WesGilster, 2017-03-05:
Click on "printers" on the dashboard page. Then you'll find the set of community printers that are available to install automatically. If none of these meet your needs you can click on the "Create" button. That will allow you to setup all of the settings you are looking for. From this point on, everything should be gui setup.

WesGilster, 2017-03-06:
I'll need two things:
1. Provide a screenshot of your printers page 
2. Click the "download" button on any page of Photonic3d and upload that zip file to this forum.

That should give us enough information to go on.

Jolly2015, 2017-03-06:
@WesGilster OK, thanks,here is the link,
https://github.com/Jolly2015/Photonic3D-error-log?files=1

Jolly2015, 2017-03-06:
Then I change  google browser and now I can create a printer on printersPage, but when I click "start",  it said "Failed to assign PrinterFirmware for Autodetect 3d printer firmware",I think the marlin firmware should be modify somewhere and I am try it now.

Jolly2015, 2017-03-06:
Everything is OK now:
1.Use chrome browser
2.Use right firmware

jmkao, 2017-03-06:
What browser were you using before that didn't work?

WesGilster, 2017-03-07:
Yeah I would like to know that as well. I test mostly with Chrome, but would like to know what problems you might be having.

Jolly2015, 2017-03-07:
@jmkao @WesGilster  I using 360 Browser before, can I change language for Photonic3D, If yes, is there any instruction?

kloknibor, 2017-03-07:
@Jolly2015 we don't have translations for photonic3D I'm afraid. However what you could do is google translate from the chrome browser on photonic :)!

Jolly2015, 2017-03-07:
I can understand that, I just saw another  country's language  version somewhere,  so I am very curious if I can ,LOL

kloknibor, 2017-03-07:
@Jolly2015 most text is hardcoded throughout everywhere in the code basically :P So at this point translating would be quite an task! But who knows maybe in the future :)!

kloknibor, 2017-03-07:
Also we had some trouble in IE11 and rewrote the function in settings.js to this (code is outdated I''m merging with Wes's code currently ) : 

```
function refreshSelectedPrinter(printerList) {
        	var foundPrinter = false;
        	if (printerList.length == 1 &&  controller.autodirect != 'disabled') {
        		controller.currentPrinter = printerList[0];
        		foundPrinter = true;
        	} else {
        		var printersStarted = 0;
        		var currPrinter = null;
	        	for (var i = 0; i < printerList.length; i++) {
				// had to change as for ___ of ____ isn't supported in IE11 :(
	        		if (printersStarted > 1) {
	        			break;
	        		}
	        		if (printerList[i].started) {
	        			printersStarted += 1;
	        			currPrinter = printerList[i];
	        		}
	        		if (controller.currentPrinter != null && printerList[i].configuration.name === controller.currentPrinter.configuration.name) {
	        			controller.currentPrinter = printerList[i];
	        			foundPrinter = true;
	        		}
	        	}
	        	if (printersStarted == 1 && controller.autodirect != 'disabled') {
	        		controller.currentPrinter = currPrinter;
	        		foundPrinter = true;
	        	}
        	}
        	if (!foundPrinter) {
        		controller.currentPrinter = null;
        	}
        }
```

WesGilster, 2017-03-07:
Google has a nice little "translate" button that they can add to your 
page. Probably the easiest route for now.

WesGilster, 2017-03-09:
Setup a pull request to my dev repo for that change and I'll take a look at it and see if I can merge it in.

WesGilster, 2017-06-28:
This code was merged in from Robin's repo fairly recently.

closed 2017-06-28 by WesGilster